micro-ROS/micro_ros_zephyr_module #1: Follow Zephyr out-of-tree module example

gregshue, 2020-09-28:
When Zephyr Project resolves https://github.com/zephyrproject-rtos/zephyr/issues/28438, update module to follow example interfacing pattern for out-of-tree modules.

protobits, 2021-06-28:
I'm trying to now add the micro ros as a proper module to my project (adding an entry to my west.yml manifest). However, it seems that this repo does not actually follow the module layout as the module is actually located inside modules/libmicroros (containing the required zephyr/module.yml file).
I think that this should be a separate repo so that it can be managed by west properly (in that case, the current repo would have its own west.yml).

gregshue, 2021-06-29:
It can still be extended to a proper Zephyr module by adding a '$repo_root/zephyr/module.yml' where this file imports the other module.yml. I don't know how many other settings in that file will also need to be explicit. Make sure to use at least west 0.11.* and Zephyr 2.6.0.

peterpolidoro, 2022-07-01:
So is the goal of this repository to be a zephyr module that only depends on Zephyr build tools, like west?

You have done a great job with making micro-ROS work on lots of platforms. It makes it a little confusing, though, when getting started, to figure out how to setup a proper build environment with all of the correct dependencies since there are so many options and repositories and versions. The Docker images really help and micro_ros_setup as well, but it would be nice if there was an option for a very minimal build environment if a user wants to use a single platform, like zephyr.

Is that the purpose of this module? If so, what still needs to be done to make this happen, just refactoring to make it have the same layout and files as the zephyr [example-application](https://github.com/zephyrproject-rtos/example-application)?

Will this zephyr module be independent of [micro_ros_setup](https://github.com/micro-ROS/micro_ros_setup)? Can it be configured just using devicetree and KConfig like normal zephyr projects without needing the micro_ros_setup shell scripts?

Will [zephyr_apps](https://github.com/micro-ROS/zephyr_apps) be able to use this zephyr module or will they need to be modified?

Will this zephyr module be independent of the ROS build tools and a ROS workspace or will those still be necessary to get this to build? Are the ROS tools only necessary if there are custom messages or something? Why is colcon necessary?

I would be happy to help, although I am still just learning zephyr and micro-ROS.