micro-ROS/micro_ros_zephyr_module #2: error : libmicroros.a(/) in archive is not an object

DenvE, 2020-10-22:
Following steps to reproduce:
environment : macos 10.15.7
installing tools : 
      python3 -m pip install colcon-common-extensions
      pip3 install catkin_pkg lark-parser empy
clone the git
in zephyr directory execute
      west init && west update
add root dir:
    export GNUARMEMB_TOOLCHAIN_PATH="$HOME/tools/gcc-arm-none-eabi-8-2018-q4-major"
    export ZEPHYR_TOOLCHAIN_VARIANT="gnuarmemb"    

source modules/libmicroros/zephyr/zephyr/zephyr-env.sh
west build -b disco_l475_iot1 micro_ros_zephyr_module

result:
  Linking C executable zephyr/zephyr_prebuilt.elf

tools/gcc-arm-none-eabi-8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: /Users/d/_Projects/gitprojects/w/micro_ros_zephyr_module/modules/libmicroros/libmicroros.a: member /Users/d/_Projects/gitprojects/w/micro_ros_zephyr_module/modules/libmicroros/libmicroros.a(/) in archive is not an object

DenvE, 2020-10-23:
Solution:

add RANLIB=${CMAKE_RANLIB} to build_command of externalproject_add
use ${RANLIAB} in stead of ranlib in libmicroros.mk file

pablogs9, 2020-10-23:
Related: https://github.com/micro-ROS/micro_ros_zephyr_module/pull/3

pablogs9, 2020-10-23:
Could you explain why is ranlib not working by default?

DenvE, 2020-10-23:
Well, in cmake you create an external project and pass along the required parameters lik AR, CC. These are known in the new project scope, and are linked to the cross compilation tool (gnu) when running cmake in the first loop.
Project is build in second loop, so ranlib is defined to the host (as not passed in the project during first loop, so referring to macos host toolchain, (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib) which is know system wide.

so generate .a file has not the gnu gcc format and can not be linked

pablogs9, 2020-10-23:
Ok, CI added, now we can check if everything works with an usual workflow: https://github.com/micro-ROS/micro_ros_zephyr_module/pull/5

pablogs9, 2020-10-23:
Closing since https://github.com/micro-ROS/micro_ros_zephyr_module/pull/3 merged

closed 2020-10-23 by pablogs9