micro-ROS/micro_ros_zephyr_module #4: undefined reference to `_fini'

DenvE, 2020-10-23:
Following steps to reproduce:
environment : macos 10.15.7
zephy: 2.4.99 
installing tools :
python3 -m pip install colcon-common-extensions
pip3 install catkin_pkg lark-parser empy
clone the git
in zephyr directory execute
west init && west update
add root dir:
export GNUARMEMB_TOOLCHAIN_PATH="$HOME/tools/gcc-arm-none-eabi-8-2018-q4-major"
export ZEPHYR_TOOLCHAIN_VARIANT="gnuarmemb"

source modules/libmicroros/zephyr/zephyr/zephyr-env.sh
west build -b disco_l475_iot1 micro_ros_zephyr_module

RANLIB patch applied!!!

gcc-arm-none-eabi-8-2018-q4-major/arm-none-eabi/lib/thumb/v7e-m/nofp/libc.a(lib_a-fini.o): in function `__libc_fini_array':
fini.c:(.text.__libc_fini_array+0x26): undefined reference to `_fini'

DenvE, 2020-10-23:
Info: in the build i see that the file /micro_ros_src/src/rosidl/rosidl_runtime_c/src/string_functions.c is compiled in and use exit(1) function.  As far as i know, for embedded applications this should not be used.

DenvE, 2020-10-23:
Hi,

the issue is solve if changing exit(???) into while(1) in following files:

libmicroros/micro_ros_src/src/rcl/rcl/src/rcl/rmw_implementation_identifier_check.c
/libmicroros/micro_ros_src/src/rosidl/rosidl_runtime_c/src/u16string_functions.c
/libmicroros/micro_ros_src/src/rosidl/rosidl_runtime_c/src/string_functions.c

Solution should be added to other repo as these file are imported from those repo's.
?How should this be taggled ?

pablogs9, 2020-10-23:
Could you please share your workflow as a command list, Zephyr version and Zephyr SDK version.

We have this demo repo building correctly here: https://github.com/micro-ROS/micro_ros_zephyr_module/pull/5

RCL and rosidl repos should not be modified

DenvE, 2020-10-23:
Ok, from the start: 

environment : macos 10.15.7

- brew install python3-colon-common-extensions
- python3 -m pip install colcon-common-extensions
- pip3 install catkin_pkg lark-parser empy

- git clone https://github.com/micro-ROS/micro_ros_zephyr_module.git
- cd micro_ros_zephyr_module/modules/libmicroros/zephyr
- west init --mr v2.4.0
- west update
- cd ../../../../ ( back to root dir )
- export GNUARMEMB_TOOLCHAIN_PATH="$HOME/tools/gcc-arm-none-eabi-8-2018-q4-major"
- export ZEPHYR_TOOLCHAIN_VARIANT="gnuarmemb"
- source micro_ros_zephyr_module/modules/libmicroros/zephyr/zephyr/zephyr-env.sh
- west build -b disco_l475_iot1 micro_ros_zephyr_module

that's it.

results in fini.c:(.text.__libc_fini_array+0x26): undefined reference to `_fini'

pablogs9, 2020-10-23:
I guess that this is a toolchain issue, currently, we only support `ZEPHYR_TOOLCHAIN_VARIANT=zephyr`.

Please check the CI workflow which is generating a working binary: https://github.com/micro-ROS/micro_ros_zephyr_module/blob/140701ca0bd584f71cd277e7757b4bbd31763bf9/.github/workflows/ci.yml#L22

If you achieve a solution for building with `gcc-arm-none-eabi-8-2018-q4-major` please PR your changes

DenvE, 2020-10-23:
ok, zephyr sdk is not supported on macos, crosstool-ng is not supported on macos anymore.
I will try to build crosstool-NG and see if i can manage it.  

The reason is following:  the crosstool-ng provide an exit implementation.

config LIBC_NEWLIB_REGISTER_FINI
    bool
    prompt "Enable finalization function registration using atexit"
    help
        Enable finalization function registration using atexit.

that is why it is building

DenvE, 2020-10-23:
sorry correct option
config LIBC_NEWLIB_LITE_EXIT
    bool
    prompt "Enable lite exit"
    help
        Enable lite exit, a size-reduced implementation of exit that doesn't
        invoke clean-up functions such as _fini or global destructors.

DenvE, 2020-10-23:
did following
create new zephyr project with exit(1)in main function,

same gcc-arm toolchain and it compiled

need to find out the compiler and linker options and see the differences.

DenvE, 2020-10-23:
coming closer

main.cpp file with main { exit(1)} compiles
main.c file with main {exit(1)} doesn't compile  
same error : .__libc_fini_array+0x26): undefined reference to `_fini'

DenvE, 2020-10-26:
Hi, small update: a **clean** build with CONFIG_NEWLIB_LIBC_NANO=y seems **not** to work.  If build first with option=n, then put option=y, trigger build again, it builds fine

i guess nano option is already defined in "colcon" environment and not know at zephyr during linking because trigger build again execute only the linking with nano option on.

pablogs9, 2020-10-26:
AFAK if I set the `CONFIG_NEWLIB_LIBC_NANO=y` it does not build: https://github.com/micro-ROS/micro_ros_zephyr_module/pull/6

Let me know if you find a solution.

adamheinrich, 2020-10-27:
Is C++ needed for Micro-ROS? It builds fine using `gcc-arm-none-eabi-8-2018-q4-major` when the C++ config is disabled:

```
CONFIG_CPLUSPLUS=n
CONFIG_LIB_CPLUSPLUS=n
```

pablogs9, 2020-10-27:
It shouldn't be required, but maybe some ROS2 package has some C++ files that should be built (not sure about if they are used). Have you tested disabling the C++ support?

DenvE, 2020-10-27:
Hi, during compiling with nano back to n, the error appears "attempt to rename spec link", no idea as i could not trace back the reason.
I had in the past also a problem on mac that it was referring to mac os files and not the cross-tool files.

Found out the only the linker is suffering for missing linker option.

I entered new MR with proper solution, update of linker flags in cmake.in file.

adamheinrich, 2020-10-27:
Hi @DenvE, for me this "attempt to rename spec link" error message happens when `CONFIG_NEWLIB_LIBC_NANO=y` is set in the project (using the same toolchain on Linux). So your fix seems to be a step in the correct way, there is something in the micro ROS build system which adds `--specs=nano.specs` to the linker flags of the uros packages.

@pablogs9 It builds, not sure if it runs. I'll try to run this example on my board and let you know.

DenvE, 2020-10-27:
@adamheinrich
could not find back where it is set.

Building with the latest change in cmake.in file gives a running environment ;>)

adamheinrich, 2020-10-27:
Hi @pablogs9,

> Have you tested disabling the C++ support?

I have tried running the firmware with the following configuration:

```
diff --git a/prj.conf b/prj.conf
index b87d42d..15b8641 100644
--- a/prj.conf
+++ b/prj.conf
@@ -1,13 +1,14 @@
 CONFIG_GPIO=y
 CONFIG_MICROROS=y
 
-CONFIG_MAIN_STACK_SIZE=25000
+# region SRAM overflowed by 5536 bytes with stack size 25000
+CONFIG_MAIN_STACK_SIZE=8192
 CONFIG_MAIN_THREAD_PRIORITY=3
 
 CONFIG_NEWLIB_LIBC=y
 CONFIG_NEWLIB_LIBC_NANO=n
-CONFIG_CPLUSPLUS=y
-CONFIG_LIB_CPLUSPLUS=y
+CONFIG_CPLUSPLUS=n
+CONFIG_LIB_CPLUSPLUS=n
 CONFIG_PTHREAD_IPC=n
 
 CONFIG_POSIX_API=y
@@ -15,15 +16,10 @@ CONFIG_NATIVE_POSIX_TIMER=y
 CONFIG_APP_LINK_WITH_POSIX_SUBSYS=y
 CONFIG_POSIX_CLOCK=y
 
-CONFIG_STDOUT_CONSOLE=y
-CONFIG_USB=y
-CONFIG_USB_DEVICE_STACK=y
-CONFIG_USB_DEVICE_PRODUCT="Zephyr micro-ROS"
-CONFIG_LOG=y
-CONFIG_USB_CDC_ACM=y
+CONFIG_PRINTK=n
+CONFIG_RING_BUFFER=y
+CONFIG_STDOUT_CONSOLE=n
 CONFIG_SERIAL=y
 CONFIG_UART_INTERRUPT_DRIVEN=y
-CONFIG_UART_LINE_CTRL=y
-CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
-CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
-CONFIG_USB_CDC_ACM_RINGBUF_SIZE=2048
+CONFIG_MICROROS_TRANSPORT_SERIAL=y
+CONFIG_MICROROS_SERIAL_PORT="2"
```
It runs well (I can see serial traffic using the `microros/micro-ros-agent:foxy serial` docker image) for both the default configuration and toolchain set to GNU Arm Embedded using this in `CMakeLists.txt`:

```
set(ZEPHYR_TOOLCHAIN_VARIANT gnuarmemb)
set(GNUARMEMB_TOOLCHAIN_PATH /path/to/gcc-arm-none-eabi-8-2018-q4-major)
```

So it looks like C++ support is not necessary to run this example and removing it from `prj.conf` would benefit people trying to build this using the gnuarmemb toolchain.

pablogs9, 2020-10-27:
This looks really nice, could you please create a PR with these changes?

Thank you very much

DenvE, 2020-10-27:
@adamheinrich:

by this change the USB is disabled, 
to have the project options correctly set there should be a dependency added in the Kconfig file of libmicroros, if you set usb serial, it should request to set CONFIG_USB_CDC_ACM, as zephyr_usb_serial.. has a device binding to it.

adamheinrich, 2020-10-27:
@DenvE yeah the diff was to demonstrate that I did not change anything significant, the important things for my argument are 

```
CONFIG_CPLUSPLUS=n
CONFIG_LIB_CPLUSPLUS=n
```

commit e26c279c (2020-10-28): Remove explicit C++ support (fixes #4) (#9)

Neither the example code nor the micro-ROS modules included use C++
and the CONFIG_CPLUSPLUS option breaks building the project with the
gnuarmemb toolchain.
closed 2020-10-28 by pablogs9