micro-ROS/micro_ros_zephyr_module #26: Uknown error while initializing publisher

protobits, 2021-06-22:
## Issue template

- Hardware description: WaveShare Core429i board (based on STM32F429IG)
- RTOS: Zephyr
- Installation type: Zephyr module + agent in docker 
- Version or commit hash: galactic (also tried with foxy)

#### Steps to reproduce the issue
I've first added support for Core429i board to Zephyr and successfully run various samples (blink LED, serial console, USB console). I then cloned this repo and built as:
```west build -b core429i micro_ros_zephyr_module -- -DBOARD_ROOT=$(pwd)/firmware```
I then started the agent with:
```docker run --privileged --net=host -v /dev:/dev microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0```
I then start a separate process in running container to inspect the system with:
```docker exec -it <container ID> bash```

#### Expected behavior
Should see the expected topic of microros publisher. Also, I should be able to see more debug information to see what is wrong (how to do that?)

#### Actual behavior
If I inspect Zephyr console I see: 

<pre>
*** Booting Zephyr OS build zephyr-v2.6.0-4-gdf0c6f08f7fe  ***
Waiting for agent connection
[00:00:00.061,000] <inf> usb_cdc_acm: Device suspended
[00:00:00.483,000] <inf> usb_cdc_acm: Device configured
Serial port connected!
Failed status on line 58: 1. Aborting.
</pre>

On the agent output I see:
<pre>
[1624371746.118977] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1624371746.119437] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1624371747.216420] info     | Root.cpp           | create_client            | create                 | client_key: 0x494F8913, session_id: 0x81
[1624371747.216586] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1229949203, address: 0
</pre>

And if I inspect the system inside the second container process I get:
<pre>
root@pop-os:/uros_ws# ros2 topic list
/parameter_events
/rosout
root@pop-os:/uros_ws# ros2 node list
/esp32_int32_publisher
root@pop-os:/uros_ws# ros2 node info /esp32_int32_publisher 
/esp32_int32_publisher
  Subscribers:

  Publishers:

  Service Servers:

  Service Clients:

  Action Servers:

  Action Clients:

root@pop-os:/uros_ws# 
</pre>

#### Additional information

During build, this is the output I get: https://gist.github.com/protobits/2d1eebba1c47feda6c03dbd2c895320e
Note that I added: `CONFIG_MICROROS_TRANSPORT_SERIAL_USB=y` and removed: `CONFIG_NATIVE_POSIX_TIMER=y` since this is not appropriate for real hardware.

protobits, 2021-06-22:
I have debugged the code and found it fails in `rmw_publisher.c:125`, but I have no idea why.

pablogs9, 2021-06-23:
Let me know if with [this branch](https://github.com/micro-ROS/micro_ros_zephyr_module/pull/27) you are able to have the publisher working.

I have found an extrange behaviour with the typesupport when it is built with the zephyr toolchain.

Are you able to build your environment using `arm-none-eabi` instead of `arm-zephyr-eabi`?

rosterloh, 2021-06-24:
I get the same issue with using zephy-sdk for building microros
`
--- stderr: rcutils
In file included from /home/rosterloh/zephyrproject/zephyr/include/posix/sys/select.h:9,
                 from /home/rosterloh/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:50,
                 from /home/rosterloh/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:61,
                 from /home/rosterloh/zephyrproject/extras/lib/microros/micro_ros_src/src/rcutils/src/allocator.c:16:
/home/rosterloh/zephyrproject/zephyr/include/net/socket_select.h:44:10: error: expected ';' before 'int'
   44 | __syscall int zsock_select(int nfds, zsock_fd_set *readfds,
      |          ^~~~
      |          ;
In file included from /home/rosterloh/zephyrproject/build/zephyr/include/generated/syscalls/socket_select.h:10,
                 from /home/rosterloh/zephyrproject/zephyr/include/net/socket_select.h:147,
                 from /home/rosterloh/zephyrproject/zephyr/include/posix/sys/select.h:9,
                 from /home/rosterloh/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:50,
                 from /home/rosterloh/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:61,
                 from /home/rosterloh/zephyrproject/extras/lib/microros/micro_ros_src/src/rcutils/src/allocator.c:16:
/home/rosterloh/zephyrproject/zephyr/include/syscall.h:93:21: error: expected ';' before '_Bool'
   93 | static ALWAYS_INLINE bool z_syscall_trap(void)
      |                     ^
      |                     ;
In file included from /home/rosterloh/zephyrproject/zephyr/include/net/socket_select.h:147,
                 from /home/rosterloh/zephyrproject/zephyr/include/posix/sys/select.h:9,
                 from /home/rosterloh/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:50,
                 from /home/rosterloh/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:61,
                 from /home/rosterloh/zephyrproject/extras/lib/microros/micro_ros_src/src/rcutils/src/allocator.c:16:
/home/rosterloh/zephyrproject/build/zephyr/include/generated/syscalls/socket_select.h:26:19: error: static declaration of 'zsock_select' follows non-static declaration
   26 | static inline int zsock_select(int nfds, zsock_fd_set * readfds, zsock_fd_set * writefds, zsock_fd_set * exceptfds, struct zsock_timeval * timeout)
      |                   ^~~~~~~~~~~~
In file included from /home/rosterloh/zephyrproject/zephyr/include/posix/sys/select.h:9,
                 from /home/rosterloh/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:50,
                 from /home/rosterloh/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:61,
                 from /home/rosterloh/zephyrproject/extras/lib/microros/micro_ros_src/src/rcutils/src/allocator.c:16:
/home/rosterloh/zephyrproject/zephyr/include/net/socket_select.h:44:15: note: previous declaration of 'zsock_select' was here
   44 | __syscall int zsock_select(int nfds, zsock_fd_set *readfds,
      |               ^~~~~~~~~~~~
gmake[3]: *** [CMakeFiles/rcutils.dir/build.make:82: CMakeFiles/rcutils.dir/src/allocator.c.obj] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:166: CMakeFiles/rcutils.dir/all] Error 2
gmake[1]: *** [Makefile:146: all] Error 2
---
`

pablogs9, 2021-06-24:
@rosterloh why is your issue related? Your log shows a building failure, @protobits's issue is building and being flashed.

@protobits Did you check the branch I mention? Does it solves the error in line `rmw_publisher.c:125`?

rosterloh, 2021-06-24:
@pablogs9 I suspect it's related you your branch fix you mentioned above and wanted to share that I'd tried it and it made no difference.

pablogs9, 2021-06-24:
Please @rosterloh open a new issue with instructions for replicating your use case.

rosterloh, 2021-06-24:
Done. See #28

protobits, 2021-06-26:
> Let me know if with [this branch](https://github.com/micro-ROS/micro_ros_zephyr_module/pull/27) you are able to have the publisher working.
> 
> I have found an extrange behaviour with the typesupport when it is built with the zephyr toolchain.

I've tried the update using galactic branch (which I see now contains the fix) and I see no difference.
 
> Are you able to build your environment using `arm-none-eabi` instead of `arm-zephyr-eabi`?

I don't know how to set zephyr to use system ARM instead of Zephyr's SDK. Is it working for you with
Zephyr's SDK? I wonder what the difference is with your setup as I'm only changing the board, which
shouldn't affect this right?

pablogs9, 2021-06-28:
Have you cleaned the library doing a`make clean` inside `modules/libmicroros`?

protobits, 2021-06-28:
Thanks, now it works. I didn't noticed there were whole other build folders inside microros folder (I was just deleting Zephyr's build/).

closed 2021-06-28 by protobits