micro-ROS/micro_ros_zephyr_module #28: Project fails to build

rosterloh, 2021-06-24:
## Issue template

- Hardware description: adafruit_feather_stm32f405
- RTOS: Zephyr
- Installation type: Ubuntu 21.04 with pip libraries as per README
- Version or commit hash: galactic

#### Steps to reproduce the issue
- Setup Zephyr as per [getting started guide](https://docs.zephyrproject.org/latest/getting_started/index.html)
- Clone micro_ros repo into same root as zephyr with git clone https://github.com/micro-ROS/micro_ros_zephyr_module micro_ros
- west build -b adafruit_feather_stm32f405 micro_ros

#### Expected behavior
micro-ROS library should build and application should link 

#### Actual behavior
micro-ROS library fails to build

#### Additional information
```bash
[143/159] Performing build step for 'libmicroros_project'
FAILED: libmicroros-prefix/src/libmicroros_project-stamp/libmicroros_project-build /home/rosterloh/zephyrproject/extras/lib/microros/libmicroros.a 
cd /home/rosterloh/zephyrproject/extras/lib/microros && make -f libmicroros.mk "X_CFLAGS=-I/home/rosterloh/zephyrproject/zephyr/include -I/home/rosterloh/zephyrproject/build/zephyr/include/generated -I/home/rosterloh/zephyrproject/zephyr/soc/arm/st_stm32/stm32f4 -I/home/rosterloh/zephyrproject/zephyr/lib/libc/newlib/include -I/home/rosterloh/zephyrproject/zephyr/drivers -I/home/rosterloh/zephyrproject/zephyr/soc/arm/st_stm32/common -I/home/rosterloh/zephyrproject/zephyr/subsys/usb -I/home/rosterloh/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/rosterloh/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/soc -I/home/rosterloh/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include -I/home/rosterloh/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/Legacy -I/home/rosterloh/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -I/home/rosterloh/zephyrproject/extras/include -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -DBUILD_VERSION=zephyr-v2.6.0-495-gf5b0f92bb718 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -DSTM32F405xx -DUSE_HAL_DRIVER -DUSE_FULL_LL_DRIVER -DCORE_CM4 -DHSE_VALUE=12000000 -Os -imacros /home/rosterloh/zephyrproject/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros /home/rosterloh/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/rosterloh/zephyrproject/extras/applications/uros=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/rosterloh/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/rosterloh/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections " "X_CXXFLAGS=-I/home/rosterloh/zephyrproject/zephyr/include -I/home/rosterloh/zephyrproject/build/zephyr/include/generated -I/home/rosterloh/zephyrproject/zephyr/soc/arm/st_stm32/stm32f4 -I/home/rosterloh/zephyrproject/zephyr/lib/libc/newlib/include -I/home/rosterloh/zephyrproject/zephyr/drivers -I/home/rosterloh/zephyrproject/zephyr/soc/arm/st_stm32/common -I/home/rosterloh/zephyrproject/zephyr/subsys/usb -I/home/rosterloh/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/rosterloh/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/soc -I/home/rosterloh/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include -I/home/rosterloh/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/Legacy -I/home/rosterloh/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -I/home/rosterloh/zephyrproject/extras/include -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -DBUILD_VERSION=zephyr-v2.6.0-495-gf5b0f92bb718 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -DSTM32F405xx -DUSE_HAL_DRIVER -DUSE_FULL_LL_DRIVER -DCORE_CM4 -DHSE_VALUE=12000000 -Os -fcheck-new -fno-exceptions -fno-rtti -imacros /home/rosterloh/zephyrproject/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros /home/rosterloh/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/rosterloh/zephyrproject/extras/applications/uros=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/rosterloh/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/rosterloh/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections " X_CC=/home/rosterloh/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc X_AR=/home/rosterloh/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-ar X_RANLIB=/home/rosterloh/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-ranlib X_CXX=/home/rosterloh/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++ COMPONENT_PATH=/home/rosterloh/zephyrproject/zephyr/../extras/lib/microros ZEPHYR_BASE=/home/rosterloh/zephyrproject/zephyr PROJECT_BINARY_DIR=/home/rosterloh/zephyrproject/build/zephyr && /usr/bin/cmake -E touch /home/rosterloh/zephyrproject/build/libmicroros-prefix/src/libmicroros_project-stamp/libmicroros_project-build
. /home/rosterloh/zephyrproject/zephyr/../extras/lib/microros/utils.sh; \
cp /home/rosterloh/zephyrproject/zephyr/../extras/lib/microros/colcon.meta /home/rosterloh/zephyrproject/zephyr/../extras/lib/microros/configured_colcon.meta; \
ZEPHYR_CONF_FILE=/home/rosterloh/zephyrproject/build/zephyr/.config; \
update_meta_from_zephyr_config "CONFIG_MICROROS_NODES" "rmw_microxrcedds" "RMW_UXRCE_MAX_NODES"; \
update_meta_from_zephyr_config "CONFIG_MICROROS_PUBLISHERS" "rmw_microxrcedds" "RMW_UXRCE_MAX_PUBLISHERS"; \
update_meta_from_zephyr_config "CONFIG_MICROROS_SUBSCRIBERS" "rmw_microxrcedds" "RMW_UXRCE_MAX_SUBSCRIPTIONS"; \
update_meta_from_zephyr_config "CONFIG_MICROROS_CLIENTS" "rmw_microxrcedds" "RMW_UXRCE_MAX_CLIENTS"; \
update_meta_from_zephyr_config "CONFIG_MICROROS_SERVERS" "rmw_microxrcedds" "RMW_UXRCE_MAX_SERVICES"; \
update_meta_from_zephyr_config "CONFIG_MICROROS_RMW_HISTORIC" "rmw_microxrcedds" "RMW_UXRCE_MAX_HISTORY"; \
update_meta_from_zephyr_config "CONFIG_MICROROS_XRCE_DDS_HISTORIC" "rmw_microxrcedds" "RMW_UXRCE_STREAM_HISTORY"; \
update_meta "microxrcedds_client" "UCLIENT_PROFILE_SERIAL=OFF"; \
update_meta "microxrcedds_client" "UCLIENT_PROFILE_UDP=OFF"; \
update_meta "microxrcedds_client" "UCLIENT_PROFILE_TCP=OFF"; \
update_meta "microxrcedds_client" "UCLIENT_PROFILE_CUSTOM_TRANSPORT=ON"; \
update_meta "microxrcedds_client" "UCLIENT_PROFILE_STREAM_FRAMING=ON"; \
update_meta "rmw_microxrcedds" "RMW_UXRCE_TRANSPORT=custom";
rm -f /home/rosterloh/zephyrproject/zephyr/../extras/lib/microros/zephyr_toolchain.cmake; \
cat /home/rosterloh/zephyrproject/zephyr/../extras/lib/microros/zephyr_toolchain.cmake.in | \
	sed "s/@CMAKE_C_COMPILER@/\/home\/rosterloh\/zephyr-sdk\/arm-zephyr-eabi\/bin\/arm-zephyr-eabi-gcc/g" | \
	sed "s/@CMAKE_CXX_COMPILER@/\/home\/rosterloh\/zephyr-sdk\/arm-zephyr-eabi\/bin\/arm-zephyr-eabi-g++/g" | \
	sed "s/@CMAKE_SYSROOT@/\/home\/rosterloh\/zephyrproject\/zephyr\/..\/extras\/lib\/microros/g" | \
	sed "s/@CFLAGS@/-c -I\/home\/rosterloh\/zephyrproject\/zephyr\/include\/posix -I\/home\/rosterloh\/zephyrproject\/build\/zephyr\/include\/generated -I\/home\/rosterloh\/zephyrproject\/zephyr\/include -I\/home\/rosterloh\/zephyrproject\/build\/zephyr\/include\/generated -I\/home\/rosterloh\/zephyrproject\/zephyr\/soc\/arm\/st_stm32\/stm32f4 -I\/home\/rosterloh\/zephyrproject\/zephyr\/lib\/libc\/newlib\/include -I\/home\/rosterloh\/zephyrproject\/zephyr\/drivers -I\/home\/rosterloh\/zephyrproject\/zephyr\/soc\/arm\/st_stm32\/common -I\/home\/rosterloh\/zephyrproject\/zephyr\/subsys\/usb -I\/home\/rosterloh\/zephyrproject\/modules\/hal\/cmsis\/CMSIS\/Core\/Include -I\/home\/rosterloh\/zephyrproject\/modules\/hal\/stm32\/stm32cube\/stm32f4xx\/soc -I\/home\/rosterloh\/zephyrproject\/modules\/hal\/stm32\/stm32cube\/stm32f4xx\/drivers\/include -I\/home\/rosterloh\/zephyrproject\/modules\/hal\/stm32\/stm32cube\/stm32f4xx\/drivers\/include\/Legacy -I\/home\/rosterloh\/zephyrproject\/modules\/hal\/stm32\/stm32cube\/common_ll\/include -I\/home\/rosterloh\/zephyrproject\/extras\/include -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -DBUILD_VERSION= [... cut ...]
Failed   <<< rcutils [2.80s, exited with code 2]
Aborted  <<< microxrcedds_client [0.77s]
Aborted  <<< rosidl_adapter [1.60s]
Aborted  <<< tracetools [0.51s]

Summary: 4 packages finished [4.53s]
  1 package failed: rcutils
  3 packages aborted: microxrcedds_client rosidl_adapter tracetools
  3 packages had stderr output: microcdr rcutils tracetools
  49 packages not processed
make: *** [libmicroros.mk:100: /home/rosterloh/zephyrproject/zephyr/../extras/lib/microros/micro_ros_src/install] Error 2
```

pablogs9, 2021-06-24:
Make sure that you are using one of the following zephyr releases: Zephyr RTOS v2.4.0 (SDK 0.11.4), v2.5.0 (SDK 0.11.4) or (v2.6.0 SDK 0.12.4).

You will be able to change the Zephyr version by doing a `git checkout` in the Zephyr repo that `west` downloads.

Which version are you using? Which SDK?

rosterloh, 2021-06-24:
Zephy 2.6.0 with SDK version 0.12.4

pablogs9, 2021-06-24:
Just for testing purposes, I have created [this PR](https://github.com/micro-ROS/micro_ros_zephyr_module/pull/29) changing the board used in the CI to yours. If this works I would recommend you to follow [those steps](https://github.com/micro-ROS/micro_ros_zephyr_module/blob/b32ebee005613572fbebbeffb282a57cf3f97a7d/.github/workflows/ci.yml#L35)

pablogs9, 2021-07-02:
Any update on this?

rosterloh, 2021-07-05:
@pablogs9 the issues I was having with build were due to me being on the main branch of zephyr instead of 2.6.0 tag. It builds with the correct tag but this issue I'm facing will come up for you the next time you update so I imagine it's worth looking into.

closed 2021-07-05 by pablogs9