micro-ROS/micro_ros_zephyr_module #34: Parameter server not working correctly

rosterloh, 2021-07-28:
## Issue template

- Hardware description: adafruit_feather_stm32f405
- RTOS: Zephyr 2.6.0
- Installation type: libmicroros as a zephyr module
- Version or commit hash: galactic

#### Steps to reproduce the issue
https://gist.github.com/rosterloh/5b66d993c62bc975d5ca6bbe0ade1fa0

#### Expected behavior
ros2 param get /gardin_ec laser_pulse_us should return 20

#### Actual behavior
*** Booting Zephyr OS build zephyr-v2.6.0  ***
Waiting for agent connection
Serial port connected!
[00:00:00.000,000] <err> dac_ltc2633: Set internal Vref failed
[00:00:00.000,000] <dbg> servo.servo_pwm_init: Servo 0 configured for full:12000 min:(705, 0), max:(2309, 12000)
[00:00:00.000,000] <dbg> servo.servo_pwm_init: Servo 1 configured for full:12000 min:(705, 0), max:(2309, 12000)
[00:00:00.000,000] <dbg> servo.servo_pwm_init: Servo 2 configured for full:18000 min:(544, 0), max:(2400, 18000)
[00:00:00.004,000] <inf> main: uROS Application 1.0.0

[00:00:00.060,000] <inf> usb_cdc_acm: Device suspended
[00:00:00.484,000] <inf> usb_cdc_acm: Device configured
INFO: rcl_wait timeout 0 ms
[00:00:02.078,000] <inf> main: Trigger delayed changed to 750
[00:00:02.084,000] <inf> main: Laser pulse width changed to 20
cannot allocate received sequence in ros_message

#### Additional information

pablogs9, 2021-07-28:
Have you isolated this? If you remove the parameter server it works as expected?

rosterloh, 2021-07-28:
It does yes. In this particular case the subscription works correctly but the server does not which is a separate issue #35

pablogs9, 2021-07-28:
Have you configured `# Modules -> micro-ROS support` in `west build -t menuconfig`, by default RMW in built with 1 entity of each type, and in this case you need at least 6 services:

[image]

rosterloh, 2021-07-28:
I have yes. I mentioned it in issue #35 but for clarity I have configured 10 of each type just to be sure

pablogs9, 2021-07-28:
I just have tested this code and it works as expected, I'm outputting the debug through a topic:

```c
#include <zephyr.h>

#include <rcl/rcl.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rcl/error_handling.h>
#include <rclc_parameter/rclc_parameter.h>

#include <std_msgs/msg/u_int16.h>
#include <std_msgs/msg/header.h>
#include <gardin_interfaces/msg/servos.h>
#include <gardin_interfaces/srv/control.h>

#include <rmw_microros/rmw_microros.h>
#include <microros_transports.h>

rcl_node_t node;
rcl_publisher_t publisher;
std_msgs__msg__Header debug_msg;
rclc_executor_t executor;
rclc_parameter_server_t param_server;
rcl_subscription_t servo_subscription;
gardin_interfaces__msg__Servos servo_msg;

void on_parameter_changed(Parameter * param)
{
	if (strcmp(param->name.data, "trigger_delay_us") == 0 && param->value.type == RCLC_PARAMETER_INT)
	{
		sprintf(debug_msg.frame_id.data, "mod: trigger_delay_us %d", param->value.integer_value);
		debug_msg.frame_id.size = strlen(debug_msg.frame_id.data);
		rcl_publish(&publisher, &debug_msg, NULL);

		//publish = param->value.bool_value;
		//// LOG_INF("Publish %s", (publish) ? "ON" : "OFF");
		// LOG_INF("Trigger delayed changed to %d", (uint16_t)param->value.integer_value);
	}
	else if (strcmp(param->name.data, "laser_pulse_us") == 0 && param->value.type == RCLC_PARAMETER_INT)
	{
		sprintf(debug_msg.frame_id.data, "mod: laser_pulse_us %d", param->value.integer_value);
		debug_msg.frame_id.size = strlen(debug_msg.frame_id.data);
		rcl_publish(&publisher, &debug_msg, NULL);

		// LOG_INF("Laser pulse width changed to %d", (uint16_t)param->value.integer_value);
	}
	sprintf(debug_msg.frame_id.data, "");
	debug_msg.frame_id.size = strlen(debug_msg.frame_id.data);

}

void servo_subscription_callback(const void * msgin)
{
	const gardin_interfaces__msg__Servos * msg = (const gardin_interfaces__msg__Servos *)msgin;
	debug_msg.stamp.sec = msg->id + msg->angle;
	rcl_publish(&publisher, &debug_msg, NULL);
	// LOG_INF("Moving Servo %d to %d", msg->id, msg->angle);
	// set_servo_angle(msg->id, msg->angle);
}

void control_service_callback(const void * req, void * res)
{
	// int err;
	gardin_interfaces__srv__Control_Request * req_in = (gardin_interfaces__srv__Control_Request *) req;
	gardin_interfaces__srv__Control_Response * res_in = (gardin_interfaces__srv__Control_Response *) res;

	// LOG_INF("Service request value: %d: %d.", req_in->module, req_in->command);

	// err = set_gpio_state(req_in->module, req_in->command);
	// res_in->success = (err == 0);
	debug_msg.stamp.sec = req_in->module + req_in->command;
	rcl_publish(&publisher, &debug_msg, NULL);
	res_in->success = req_in->module + req_in->command;
}

void main(void)
{
	int err;
	rcl_ret_t rc;

	// LOG_INF("uROS Application %s\n", APP_VERSION_STR);

	// err = board_init();
	// if (err) {
	// 	// LOG_ERR("board init failed (err %d)\n", err);
	// 	return;
	// }

	rmw_uros_set_custom_transport(
		MICRO_ROS_FRAMING_REQUIRED,
		(void *) &default_params,
		zephyr_transport_open,
		zephyr_transport_close,
		zephyr_transport_write,
		zephyr_transport_read
	);

	rcl_allocator_t allocator = rcl_get_default_allocator();

	rclc_support_t support;
	rc = rclc_support_init(&support, 0, NULL, &allocator);
	if (rc != RCL_RET_OK) {
		// LOG_ERR("Failed to init support (%d)", rc);
		return;
	}

	rc = rclc_node_init_default(&node, "gardin_ec", "", &support);
	if (rc != RCL_RET_OK) {
		// LOG_ERR("Failed to create node (%d)", rc);
		goto error;
	}

	rclc_publisher_init_default(
		&publisher,
		&node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Header),
		"debug_pub");

	debug_msg.frame_id.data = malloc(100);
	sprintf(debug_msg.frame_id.data, "");
	debug_msg.frame_id.capacity = 100;
	debug_msg.frame_id.size = 0;

	// Max params 4 by default. Do init_with_options
	rc = rclc_parameter_server_init_default(&param_server, &node);
	if (rc != RCL_RET_OK) {
		// LOG_ERR("Failed to create parameter server (%d)", rc);
	}

	rcl_service_t service;
	rc = rclc_service_init_default(
		&service,
		&node,
		ROSIDL_GET_SRV_TYPE_SUPPORT(gardin_interfaces, srv, Control),
		"/control");
	if (rc != RCL_RET_OK) {
		// LOG_ERR("Failed to create control service (%d)", rc);
		goto error;
	}

	rc = rclc_subscription_init_default(
		&servo_subscription,
		&node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(gardin_interfaces, msg, Servos),
		"/servo");
	if (rc != RCL_RET_OK) {
		// LOG_ERR("Failed to create servo subscription (%d)", rc);
		goto error;
	}

	// Parameter server has 5 services and 1 publisher */
	rc = rclc_executor_init(&executor, &support.context, 8, &allocator);
	if (rc != RCL_RET_OK) {
		// LOG_ERR("Failed to init executor (%d)", rc);
		goto error;
	}

	rc = rclc_executor_add_parameter_server(&executor, &param_server, on_parameter_changed);
	if (rc != RCL_RET_OK) {
		// LOG_ERR("Failed to init parameter service (%d)", rc);
		goto error;
	}

	gardin_interfaces__srv__Control_Request req;
	gardin_interfaces__srv__Control_Response res;
	rc = rclc_executor_add_service(&executor, &service, &req, &res, control_service_callback);
	if (rc != RCL_RET_OK) {
		// LOG_ERR("Failed to init service (%d)", rc);
		goto error;
	}

	rc = rclc_executor_add_subscription(
		&executor,
		&servo_subscription,
		&servo_msg,
		&servo_subscription_callback,
		ON_NEW_DATA);
	if (rc != RCL_RET_OK) {
		// LOG_ERR("Failed to init servo subscription (%d)", rc);
		goto error;
	}

	rclc_add_parameter(&param_server, "trigger_delay_us", RCLC_PARAMETER_INT);
	rclc_add_parameter(&param_server, "laser_pulse_us", RCLC_PARAMETER_INT);

	rclc_parameter_set_int(&param_server, "trigger_delay_us", 750);
	rclc_parameter_set_int(&param_server, "laser_pulse_us", 20);

	rclc_executor_spin(&executor);

error:
	rc = rcl_service_fini(&service, &node);
	if (rc != RCL_RET_OK) {
		// LOG_ERR("Failed deinit service (%d)", rc);
	}

	rc = rcl_subscription_fini(&servo_subscription, &node);

	rc = rcl_node_fini(&node);
	if (rc != RCL_RET_OK) {
		// LOG_ERR("Failed deinit node (%d)", rc);
	}
}
```

pablogs9, 2021-07-28:
How many stack and heap are you configuring for you Zephyr task?

pablogs9, 2021-07-28:
Can you share the output of your micro-ROS Agent using the flag `-v6`? Also, how are you instantiating the agent?

rosterloh, 2021-07-28:
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:galactic serial --dev /dev/ttyACM1 -v6
```
[1627484521.501298] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1627484521.501484] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1627484522.551317] info     | Root.cpp           | create_client            | create                 | client_key: 0x2ACC1547, session_id: 0x81
[1627484522.551367] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2ACC1547, address: 0
[1627484522.551411] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2ACC1547, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1627484522.552537] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x2ACC1547, len: 40, data: 
0000: 81 80 00 00 01 07 20 00 00 0A 00 01 01 03 00 00 12 00 00 00 00 01 00 00 0A 00 00 00 67 61 72 64
0020: 69 6E 5F 65 63 00 00 00
[1627484522.561816] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x2ACC1547, participant_id: 0x000(1)
[1627484522.561888] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2ACC1547, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1627484522.561926] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2ACC1547, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1627484522.563133] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x2ACC1547, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1627484522.573994] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x2ACC1547, len: 244, data: 
0000: 81 80 01 00 01 07 EB 00 00 0B 00 08 08 03 00 00 DD 00 00 00 1A 00 00 00 2F 67 61 72 64 69 6E 5F
0020: 65 63 2F 67 65 74 5F 70 61 72 61 6D 65 74 65 72 73 00 00 20 32 00 00 00 72 63 6C 5F 69 6E 74 65
0040: 72 66 61 63 65 73 3A 3A 73 72 76 3A 3A 64 64 73 5F 3A 3A 47 65 74 50 61 72 61 6D 65 74 65 72 73
0060: 5F 52 65 71 75 65 73 74 5F 00 00 00 33 00 00 00 72 63 6C 5F 69 6E 74 65 72 66 61 63 65 73 3A 3A
0080: 73 72 76 3A 3A 64 64 73 5F 3A 3A 47 65 74 50 61 72 61 6D 65 74 65 72 73 5F 52 65 73 70 6F 6E 73
00A0: 65 5F 00 01 23 00 00 00 72 71 2F 67 61 72 64 69 6E 5F 65 63 2F 67 65 74 5F 70 61 72 61 6D 65 74
00C0: 65 72 73 52 65 71 75 65 73 74 00 01 21 00 00 00 72 72 2F 67 61 72 64 69 6E 5F 65 63 2F 67 65 74
00E0: 5F 70 61 72 61 6D 65 74 65 72 73 52 65 70 6C 79 00 00 01 00
[1627484522.574603] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0x2ACC1547, requester_id: 0x000(7), participant_id: 0x000(1)
[1627484522.574640] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2ACC1547, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 08 00 00
[1627484522.574668] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2ACC1547, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1627484522.579313] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x2ACC1547, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1627484522.595475] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x2ACC1547, len: 288, data: 
0000: 81 80 02 00 08 01 10 00 00 0C 00 08 80 00 00 01 FF FF 00 00 00 00 00 00 01 07 04 01 00 0D 00 18
0020: 08 03 00 00 F6 00 00 00 1F 00 00 00 2F 67 61 72 64 69 6E 5F 65 63 2F 67 65 74 5F 70 61 72 61 6D
0040: 65 74 65 72 5F 74 79 70 65 73 00 20 36 00 00 00 72 63 6C 5F 69 6E 74 65 72 66 61 63 65 73 3A 3A
0060: 73 72 76 3A 3A 64 64 73 5F 3A 3A 47 65 74 50 61 72 61 6D 65 74 65 72 54 79 70 65 73 5F 52 65 71
0080: 75 65 73 74 5F 00 00 00 37 00 00 00 72 63 6C 5F 69 6E 74 65 72 66 61 63 65 73 3A 3A 73 72 76 3A
00A0: 3A 64 64 73 5F 3A 3A 47 65 74 50 61 72 61 6D 65 74 65 72 54 79 70 65 73 5F 52 65 73 70 6F 6E 73
00C0: 65 5F 00 01 28 00 00 00 72 71 2F 67 61 72 64 69 6E 5F 65 63 2F 67 65 74 5F 70 61 72 61 6D 65 74
00E0: 65 72 5F 74 79 70 65 73 52 65 71 75 65 73 74 00 01 00 00 00 26 00 00 00 72 72 2F 67 61 72 64 69
0100: 6E 5F 65 63 2F 67 65 74 5F 70 61 72 61 6D 65 74 65 72 5F 74 79 70 65 73 52 65 70 6C 79 00 00 01
[1627484522.595966] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0x2ACC1547, requester_id: 0x001(7), participant_id: 0x000(1)
[1627484522.596109] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2ACC1547, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0D 00 18 00 00
[1627484522.596176] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2ACC1547, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1627484522.596779] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x2ACC1547, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[... cut ...]
[1627484596.490418] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2ACC1547, len: 14, data: 
0000: 81 80 1A 00 05 01 06 00 00 2E 00 01 00 00
[1627484596.490427] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x2ACC1547, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 1C 00 00 00 80
[1627484596.491306] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x2ACC1547, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 1B 00 00 00 80

```

rosterloh, 2021-07-28:
CONFIG_MICROROS=y
CONFIG_MICROROS_TRANSPORT_SERIAL_USB=y
CONFIG_MICROROS_NODES="1"
CONFIG_MICROROS_PUBLISHERS="10"
CONFIG_MICROROS_SUBSCRIBERS="10"
CONFIG_MICROROS_CLIENTS="10"
CONFIG_MICROROS_SERVERS="10"

CONFIG_MAIN_STACK_SIZE=25000
CONFIG_MAIN_THREAD_PRIORITY=3

CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_NANO=n
CONFIG_PTHREAD_IPC=n

CONFIG_POSIX_API=y
CONFIG_APP_LINK_WITH_POSIX_SUBSYS=y
CONFIG_POSIX_CLOCK=y

CONFIG_STDOUT_CONSOLE=y
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_VID=0x0483
CONFIG_USB_DEVICE_PID=0x5710
CONFIG_USB_DEVICE_MANUFACTURER="Gardin"
CONFIG_USB_DEVICE_PRODUCT="Zephyr micro-ROS"
CONFIG_LOG=y
CONFIG_USB_CDC_ACM=y
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_LINE_CTRL=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
CONFIG_USB_CDC_ACM_RINGBUF_SIZE=2048

pablogs9, 2021-07-29:
It seems that your entities are being destroyed in the agent so, anyhow your code is reaching [stop flag](https://gist.github.com/rosterloh/5b66d993c62bc975d5ca6bbe0ade1fa0#file-main-c-L166). Can you check where is it going to it?

I see in the agent output that your parameter server is alerting on changes of parameters, that means that the code is reaching [your parameter set](https://gist.github.com/rosterloh/5b66d993c62bc975d5ca6bbe0ade1fa0#file-main-c-L161).

I have seen in some boards that `rclc_executor_spin` returns, have you tried something like:

```c
...
while(1){
  rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
  usleep(10000);
}
...

rosterloh, 2021-07-29:
Thanks so much for your help so far. Unfortunately this has made no difference. I've updated the gist with the changes.
```
*** Booting Zephyr OS build zephyr-v2.6.0  ***
[00:00:00.000,000] <err> dac_ltc2633: Set internal Vref failed
[00:00:00.000,000] <dbg> servo.servo_pwm_init: Servo 0 configured for full:12000 min:(705, 0), max:(2309, 12000)
[00:00:00.000,000] <dbg> servo.servo_pwm_init: Servo 1 configured for full:12000 min:(705, 0), max:(2309, 12000)
[00:00:00.000,000] <dbg> servo.servo_pwm_init: Servo 2 configured for full:18000 min:(544, 0), max:(2400, 18000)
[00:00:00.004,000] <inf> main: uROS Application 1.0.0

[00:00:00.004,000] <iWaiting for agent connection
nf> thread_analyzer: Thread analyze:
[00:00:00.004,000] <inf> thread_analyzer:  thread_analyzer     : STACK: unused 896 usage 128 / 1024 (12 %); CPU: 0 %
[00:00:00.004,000] <inf> thread_analyzer:  sysworkq            : STACK: unused 864 usage 160 / 1024 (15 %); CPU: 0 %
[00:00:00.005,000] <inf> thread_analyzer:  usbworkq            : STACK: unused 864 usage 160 / 1024 (15 %); CPU: 0 %
[00:00:00.005,000] <inf> thread_analyzer:  logging             : STACK: unused 664 usage 104 / 768 (13 %); CPU: 0 %
[00:00:00.005,000] <inf> thread_analyzer:  <log_strdup alloc failed>: STACK: unused 272 usage 48 / 320 (15 %); CPU: 0 %
[00:00:00.008,000] <inf> thread_analyzer:  <log_strdup alloc failed>: STACK: unused 23360 usage 1640 / 25000 (6 %); CPU: 99 %
[00:00:00.060,000] <inf> usb_cdc_acm: Device suspended
[00:00:00.476,000] <inf> usb_cdc_acm: Device configured
[00:01:00.009,000] <inf> thread_analyzer: Thread analyze:
[00:01:00.009,000] <inf> thread_analyzer:  thread_analyzer     : STACK: unused 776 usage 248 / 1024 (24 %); CPU: 0 %
[00:01:00.009,000] <inf> thread_analyzer:  sysworkq            : STACK: unused 864 usage 160 / 1024 (15 %); CPU: 0 %
[00:01:00.009,000] <inf> thread_analyzer:  usbworkq            : STACK: unused 864 usage 160 / 1024 (15 %); CPU: 0 %
[00:01:00.009,000] <inf> thread_analyzer:  logging             : STACK: unused 264 usage 504 / 768 (65 %); CPU: 1 %
[00:01:00.009,000] <inf> thread_analyzer:  <log_strdup alloc failed>: STACK: unused 272 usage 48 / 320 (15 %); CPU: 98 %
[00:01:00.013,000] <inf> thread_analyzer:  <log_strdup alloc failed>: STACK: unused 23360 usage 1640 / 25000 (6 %); CPU: 0 %
Serial port connected!
[00:01:23.039,000] <inf> main: Trigger delayed changed to 750
[00:01:23.046,000] <inf> main: Laser pulse width changed to 20
cannot allocate received sequence in ros_message
```

```
[1627548900.250316] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1627548900.250515] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1627548901.333952] info     | Root.cpp           | create_client            | create                 | client_key: 0x526CF23A, session_id: 0x81
[1627548901.334093] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x526CF23A, address: 0
[1627548901.334260] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x526CF23A, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1627548901.334940] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x526CF23A, len: 40, data: 
0000: 81 80 00 00 01 07 20 00 00 0A 00 01 01 03 00 00 12 00 00 00 00 01 AA AA 0A 00 00 00 67 61 72 64
0020: 69 6E 5F 65 63 00 00 00
[1627548901.362113] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x526CF23A, participant_id: 0x000(1)
[1627548901.362330] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x526CF23A, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1627548901.362460] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x526CF23A, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1627548901.364762] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x526CF23A, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1627548901.374317] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x526CF23A, len: 244, data: 
0000: 81 80 01 00 01 07 EB 00 00 0B 00 08 08 03 00 00 DD 00 00 00 1A 00 00 00 2F 67 61 72 64 69 6E 5F
0020: 65 63 2F 67 65 74 5F 70 61 72 61 6D 65 74 65 72 73 00 00 20 32 00 00 00 72 63 6C 5F 69 6E 74 65
0040: 72 66 61 63 65 73 3A 3A 73 72 76 3A 3A 64 64 73 5F 3A 3A 47 65 74 50 61 72 61 6D 65 74 65 72 73
0060: 5F 52 65 71 75 65 73 74 5F 00 00 00 33 00 00 00 72 63 6C 5F 69 6E 74 65 72 66 61 63 65 73 3A 3A
0080: 73 72 76 3A 3A 64 64 73 5F 3A 3A 47 65 74 50 61 72 61 6D 65 74 65 72 73 5F 52 65 73 70 6F 6E 73
00A0: 65 5F 00 01 23 00 00 00 72 71 2F 67 61 72 64 69 6E 5F 65 63 2F 67 65 74 5F 70 61 72 61 6D 65 74
00C0: 65 72 73 52 65 71 75 65 73 74 00 01 21 00 00 00 72 72 2F 67 61 72 64 69 6E 5F 65 63 2F 67 65 74
00E0: 5F 70 61 72 61 6D 65 74 65 72 73 52 65 70 6C 79 00 00 01 00
[1627548901.375971] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0x526CF23A, requester_id: 0x000(7), participant_id: 0x000(1)
[1627548901.376095] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x526CF23A, len: 14, data: 
[... cut ...]
[1627548927.270381] debug    | Replier.cpp        | write                    | [** <<DDS>> **]        | client_key: 0x00000003, len: 76, data: 
0000: D5 79 10 01 42 96 B2 F7 E6 07 BC CD 00 00 08 03 00 00 00 00 01 00 00 00 02 00 00 00 11 00 00 00
0020: 74 72 69 67 67 65 72 5F 64 65 6C 61 79 5F 75 73 00 00 00 00 0F 00 00 00 6C 61 73 65 72 5F 70 75
0040: 6C 73 65 5F 75 73 00 73 00 00 00 00
[1627548927.270488] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x526CF23A, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 12 00 00 00 80
```

pablogs9, 2021-07-29:
How much heap do you available for dynamic allocations? It seems that the rclc parameter server is not able to assign enough memory to its inner members so it complains about not having memory for deserializing the sequences.

rosterloh, 2021-07-29:
I've currently just used the default which is 1024. I've just tried doubling it to 2048 and have the same issue.

pablogs9, 2021-07-29:
I think that you will need a bit more, check with at least 20 kB. 
If there is a problem with the quantity of heap available [this patch](https://github.com/ros2/rclc/pull/169/files) should help.

rosterloh, 2021-07-29:
I've tried it with 20kB as well as 60kB and get the same result. I don't understand how you said it was working as expected for you. What config did you use?

pablogs9, 2021-07-29:
Here you can see my board with the code working: https://drive.google.com/file/d/1LQHPPcJgYacfcN-gocSJ8HWbd3aCOM_B/view?usp=sharing

My code: [main.c](https://github.com/micro-ROS/micro_ros_zephyr_module/files/6899847/main.c.txt)

My conf: [.config](https://github.com/micro-ROS/micro_ros_zephyr_module/files/6899856/config.txt)

rosterloh, 2021-07-29:
Thanks for this. I've been through the config and there is nothing fundamentally different in there. In fact you've got your CONFIG_HEAP_MEM_POOL_SIZE set to the default 1024. Do you think it's possible there could be something wrong with the fact I'm running the ros2 commands from another [docker](https://github.com/micro-ROS/micro_ros_zephyr_module/files/6900212/Dockerfile.txt) container?

pablogs9, 2021-07-29:
Have you modified your default middleware to FastDDS or you are using CycloneDDS (default in galactic)?

pablogs9, 2021-07-29:
Also if you are working in dockers is interesting to set some FastDDS profiles like [this](https://github.com/micro-ROS/docker/blob/cdb59b4c2b72d108e2564296d06da3a5c173d362/micro-ROS-Agent/Dockerfile#L13) with [this file]( https://github.com/micro-ROS/docker/blob/galactic/micro-ROS-Agent/disable_fastdds_shm.xml)

rosterloh, 2021-07-29:
That's it! Wrong DDS implementation. Stupid me. Thank you
```
ros@RO-PC:/workspaces/gardin_ros2$ ros2 param dump /gardin_ec --print
/gardin_ec:
  ros__parameters:
    laser_pulse_us: 20
    trigger_delay_us: 750
```

pablogs9, 2021-07-29:
Cool! I'm closing do not hesitate to reopen if you have other problem.

closed 2021-07-29 by pablogs9