micro-ROS/micro_ros_zephyr_module #35: Service calls don't work correctly with custom services

rosterloh, 2021-07-28:
- Hardware description: adafruit_feather_stm32f405
- RTOS: Zephyr 2.6.0
- Installation type: libmicroros as a zephyr module
- Version or commit hash: galactic

#### Steps to reproduce the issue
https://gist.github.com/rosterloh/5b66d993c62bc975d5ca6bbe0ade1fa0

#### Expected behavior
ros2 service call /control gardin_interfaces/srv/Control
waiting for service to become available...
requester: making request: gardin_interfaces.srv.Control_Request(module=0, command=0)

#### Actual behavior
[00:00:07.571,000] <inf> main: Service request value: 193: 202.
[00:00:15.858,000] <inf> main: Service request value: 10: 72.
[00:00:59.142,000] <inf> main: Service request value: 36: 203.
[00:01:01.874,000] <inf> main: Service request value: 195: 121.

pablogs9, 2021-07-29:
Can we close also this?

rosterloh, 2021-07-29:
Yes, verified working after adding RMW_IMPLEMENTATION=rmw_fastrtps_cpp. See #34

closed 2021-07-29 by rosterloh