micro-ROS/micro_ros_zephyr_module #41: Question: Build failure in unique_identifier_msgs

ottojo, 2021-11-08:
Hi! I'm trying to integrate micro-ros into an existing zephyr project using this module.

The build fails, and it seems like what's happening is that it tries to build message-definitions for fastrtps, which is not possible due to a `fastcdr` only being installed for "64bit", and i'm building for my ARM microcontroller.

(Why does this even happen? If i understand correctly ROS2 galactic uses cyclone exclusively, and micro-ros doesnt use either of those...)

I would appreciate if anyone has experienced this before and knows a solution, or has ideas on how to solve this.

```
CMake Error at /opt/ros/galactic/share/rosidl_typesupport_fastrtps_c/cmake/rosidl_typesupport_fastrtps_c-extras.cmake:5 (find_package):
  Could not find a configuration file for package "fastcdr" that is
  compatible with requested version "".

  The following configuration files were considered but not accepted:

    /opt/ros/galactic/lib/cmake/fastcdr/fastcdr-config.cmake, version: 1.0.20 (64bit)

Call Stack (most recent call first):
  /opt/ros/galactic/share/rosidl_typesupport_fastrtps_c/cmake/rosidl_typesupport_fastrtps_cConfig.cmake:41 (include)
  /home/jonasotto/io-board-software/modules/libmicroros/micro_ros_src/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /home/jonasotto/io-board-software/modules/libmicroros/micro_ros_src/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /home/jonasotto/io-board-software/modules/libmicroros/micro_ros_src/install/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /home/jonasotto/io-board-software/modules/libmicroros/micro_ros_src/install/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

---
Failed   <<< unique_identifier_msgs [3.52s, exited with code 1]
```

Some relevant information:
* board: teensy41
* zephyr version: mostly up-to-date main branch (our fork with a few additional drivers)
* ROS2 galactic

pablogs9, 2021-11-10:
Could you provide steps for replicating?

You should be mixing workspace... when using this module you should not have any ROS 2 workspace sourced.

ottojo, 2021-11-10:
> when using this module you should not have any ROS 2 workspace sourced

this turned out to be (one of) the problem(s), thank you!

I was also getting linker errors in the test_interface_files package (not finding typesupport functions), but that was resolved by setting `STATIC_ROSIDL_TYPESUPPORT_C=rosidl_typesupport_microxrcedds_c`.
I will report back if i stumble upon this again, in an isolated, reproducable manner (which is not plagued by a ton of issues relating to integration into an existing project...).

As this doesnt really seem to be an issue with this module, i'm going to close this, altough not having ROS2 sourced may be something worth mentioning in build-instructions (in my case, sourcing /opt/ros/... was automated, as it's usually needed...).

closed 2021-11-10 by ottojo