micro-ROS/micro_ros_zephyr_module #42: Issue :  "topic [/microROS/ping] does not appear to be published yet Could not determine the type for the passed topic"

sachinjairam, 2022-01-06:
Hai, I am trying ping_pong example with Nucleo-G474RE board. I have added 'nucleo_g474re' to the CMakeLists.txt file.

After building and flashing the firmware, created the microROS agent and built. Then I run the agent and it shows in the first figure. 

Then in a new terminal, I tried to echo the microROS topic using the following command,
        'ros2 topic echo /microROS/ping' 

But it shows the error 
    "topic [/microROS/ping] does not appear to be published yet Could not determine the type for the passed topic"
This shows in the second figure.

I tried it with both serial and serial-usb transport.
I tried it by adding '-v6' to agent run command.
I tried it by setting ROS_DOMAIN_ID=0 in all terminals.
But the error is not resolved.
Please advise me how to solve this problem.

Thank you.
Sachin

[image]
[image]

rosterloh, 2022-01-06:
I get the same issue which I'm busy debugging at the moment. Basically the RX interrupt is never triggered and so you just get the 10 retries from rclc_support_init and then it fails. I've verified that the hardware works bidirectionally enabling the shell on the CDC_ACM_0 (in my case) interface and can send and receive successfully. If I but a breakpoint [here](https://github.com/micro-ROS/micro_ros_zephyr_module/blob/galactic/modules/libmicroros/microros_transports/serial-usb/microros_transports.c#L30) it never triggers.

rosterloh, 2022-01-06:
@sachinjairam After a bit of digging and testing on and STM32F4 device which worked I noticed that both your G4 device and my L4 device use the st_stm32_usb and the F4 uses the  st_stm32_otgfs driver. For me if I remove the calls to set DCD and DSR my device communicates with the agent.

sachinjairam, 2022-01-07:
@rosterloh Thanks for your reply. I am new to ROS and ST microcontroller, so I didn't understand some terms 'DCD', 'DSR' etc. that you mentioned in the comment:
"For me if I remove the calls to set DCD and DSR my device communicates with the agent."
Can you please tell me what it means?

I have tried using '/dev/ttyACM0', but I got the same previous error. 

Thank you.
Sachin

pablogs9, 2022-01-07:
Hello @rosterloh, is this a matter of micro-ROS transports? Is there any improvement that we can do in the [transports](https://github.com/micro-ROS/micro_ros_zephyr_module/tree/galactic/modules/libmicroros/microros_transports) that eases the usage of this component?

Thanks

rosterloh, 2022-01-07:
@sachinjairam it works for me if you comment out lines [91-99](https://github.com/micro-ROS/micro_ros_zephyr_module/blob/galactic/modules/libmicroros/microros_transports/serial-usb/microros_transports.c#L91..L99)

@pablogs9 I think the transports you've provided are fine for a start but any project of decent complexity I'd say people would have to modify the transport for themselves. I'd say the biggest improvement that could be made would be to connect the failure messages from rclc into zephyr logging so that you can see the reason for failure when something goes wrong without having to step debug.

sachinjairam, 2022-01-07:
Hello @rosterloh, Thank you for your reply. It works for me also. Now my device can communicate with the agent. Thank you for your help.

Thank you.
Sachin

closed 2022-01-07 by sachinjairam