micro-ROS/micro_ros_zephyr_module #65: Error code 1 on calling `rclc_node_init_default`

aditya2592, 2022-07-28:
## Issue template
Hardware description: Nucleo F746ZG
RTOS: Zephyr 3.1.99, Zephyr SDK 0.14.0
Installation type: micro_ros_zephyr_module
Version or commit hash: humble
Transport: UDP IPv4
Agent: microros/micro-ros-agent:humble docker image

#### Steps to reproduce the issue
The issue is seen when the agent is already running on the host PC and the microROS device is started after. The microROS code first pings the agent to identify its presence and then proceeds to creating microROS entities.

1. Start microROS agent on host PC
2. Reset STM32 Nucleo device

#### Expected behavior
No error codes on calling `rclc_node_init_default`

#### Actual behavior
Error code 1 on `RCCHECK(rclc_node_init_default(&node, "zephyr_benchmark", "", &support))`. 

#### Additional information

Block of relevant code used
```
rmw_uros_set_custom_transport(MICRO_ROS_FRAMING_REQUIRED, static_cast<void*>(&default_params), zephyr_transport_open,
                                zephyr_transport_close, zephyr_transport_write, zephyr_transport_read);
  while (true) {
    while (rmw_uros_ping_agent(ping_timeout, 2) != RMW_RET_OK) {
      LOG_ERR("Wait for agent...");
    }
    LOG_INF("Agent connected.");
    allocator = rcl_get_default_allocator();
    RCCHECK(rclc_support_init(&support, 0, nullptr, &allocator))
    RCCHECK(rclc_node_init_default(&node, "zephyr_benchmark", "", &support))
    
  }

```

pablogs9, 2022-07-29:
Could you provide complete code for replicating?

I see a forever loop here, are you destroying the entities on the client side when the agent is no longer available?
Check [this approach](https://github.com/micro-ROS/micro_ros_arduino/blob/humble/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino).

pablogs9, 2022-09-19:
Any update on this?

nauticus-stu, 2022-11-27:
I get the same behavior using the UDP transport. I just ran the example targeting a board with an ethernet interface and making the appropriate changes to the `zephyr_transport_params_t` struct.

I was using Zephyr 3.2.0 and had to make some additional changes to the includes.

Debugging it I see `UXR_STATUS_ERR_DDS_ERROR` after the initial handshake with the agent.

aditya2592, 2022-11-27:
Sorry I forgot to update earlier. We are destroying entities and then init them again. The resolution we adopted for now is to just init in a loop until it succeeds. Usually it succeeds in 1-2 attempts.

nauticus-stu, 2022-12-01:
I've narrowed the problem down to the agent. If I build the agent from source everything works as expected. The packaged agents don't seem to work. I'm guessing this issue can be closed, I don't think its a microros issue.

closed 2022-12-01 by pablogs9

Hubery-wl, 2023-08-15:
I have the same problem. May I ask how you solved it? Looking forward to your reply.