micro-ROS/micro_ros_zephyr_module #73: Build issue with Zephyr 3.1

granzscientific, 2022-08-11:
Hi, I'm having a build issue with the rcutils package after doing a clean zephyr workspace setup and then cloning the repo:

I'm on Ubuntu 20.04, with zephyr-sdk 0.14.2
West and cmake versions are shown at the bottom...

Any guidance would be highly appreciated.  I've been able to successfully build Zephyr+MicroROS using the micro_ros_setup ROS2 package, but I'd prefer to use the zephyr module since I have a lot of existing zephyr-based applications already.

Starting >>> rcl_logging_interface
--- stderr: rosidl_runtime_c
CMake Warning at /home/chris.granz/zephyrproject/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /home/chris.granz/zephyrproject/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

CMake Error at /home/chris.granz/zephyrproject/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install/share/rcutils/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/home/chris.granz/zephyrproject/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install/share/rcutils/cmake/rcutilsExport.cmake'
Call Stack (most recent call first):
  /home/chris.granz/zephyrproject/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

---
Failed   <<< rosidl_runtime_c [0.46s, exited with code 1]
Aborted  <<< rcl_logging_interface [0.43s]
Aborted  <<< tracetools_read [3.54s]
Aborted  <<< tracetools_trace [3.66s]
Aborted  <<< rosidl_cli [3.81s]
Aborted  <<< microxrcedds_client [1.63s]

Summary: 6 packages finished [5.94s]
  1 package failed: rosidl_runtime_c
  5 packages aborted: microxrcedds_client rcl_logging_interface rosidl_cli tracetools_read tracetools_trace
  5 packages had stderr output: microcdr microxrcedds_client rcl_logging_interface rcutils rosidl_runtime_c
  50 packages not processed
make: *** [libmicroros.mk:103: /home/chris.granz/zephyrproject/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install] Error 1
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/chris.granz/zephyrproject/build

chris.granz@dell-fd6b163:~/zephyrproject$ west --version
West version: v0.13.1

chris.granz@dell-fd6b163:~/zephyrproject$ cmake --version
cmake version 3.24.0

pablogs9, 2022-08-12:
Try to use CMake 3.22.1 (the version installed by default in Ubuntu 22.04). We are having some issues with newer versions.

Related https://github.com/micro-ROS/micro_ros_mbed/actions/runs/2844490547

robodrome, 2022-08-12:
I am having issues as well. I am not sure what it causing it.

## System
- cmake version 3.22.1 (tested 3.24 as well, with same result)
- West version: v0.13.1
- ROS Humble (foxy gives docutils/sphinx issues)
- Ubuntu 22.04
- docker container
- zephyr 3.1
- zephyr SDK 0.14.2

## Steps to reproduce

Dockerfile:
[here](https://github.com/micro-ROS/micro_ros_zephyr_module/files/9314883/Dockerfile.txt)

Build and run:
```bash
docker build -t robodrome/zephyr-build:humble .
docker run -ti --rm robodrome/zephyr-build:humble
```

Inside container:
```bash
cd ~
git clone -b humble https://github.com/micro-ROS/micro_ros_zephyr_module.git
west build -b disco_l475_iot1 micro_ros_zephyr_module
```

## Expected outcome:
No critical errors.

## Outcome:
[here](https://github.com/micro-ROS/micro_ros_zephyr_module/files/9314877/outcome.txt)
### CMake Warnings (multiple)
```text
CMake Warning at /home/zephyr/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
```
### Cmake Errors (single)
```text
CMake Error at /opt/ros/humble/share/rosidl_typesupport_fastrtps_c/cmake/rosidl_typesupport_fastrtps_c-extras.cmake:5 (find_package):
  Could not find a configuration file for package "fastcdr" that is
  compatible with requested version "".

  The following configuration files were considered but not accepted:

    /opt/ros/humble/lib/cmake/fastcdr/fastcdr-config.cmake, version: 1.0.24 (64bit)

Call Stack (most recent call first):
  /opt/ros/humble/share/rosidl_typesupport_fastrtps_c/cmake/rosidl_typesupport_fastrtps_cConfig.cmake:41 (include)
  /home/zephyr/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /home/zephyr/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /home/zephyr/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /home/zephyr/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)
```

pablogs9, 2022-08-12:
Warnings can be ignored.

The error is related to have ROS 2 sourced when building micro-ROS. Make sure that the environment where you build micro-ROS **do not have** a ROS 2 distro sourced. Probably this is done automatically by `ros:humble` image, just check the .bashrc

robodrome, 2022-08-12:
> Warnings can be ignored.
> 
> The error is related to have ROS 2 sourced when building micro-ROS. Make sure that the environment where you build micro-ROS **do not have** a ROS 2 distro sourced. Probably this is done automatically by `ros:humble` image, just check the .bashrc

I will alter the Dockerfile and share the results. Is there a manual where I could have read that? I am missing some crucial information it seems. Thanks!

robodrome, 2022-08-12:
@granzscientific @pablogs9

Thanks Pablo, it works now. I will share my dockerfile and instructions. Hope it helps others.

## Docker
**NOTE:** ROS2 Humble on Ubuntu 22.04 Jammy

### Build
Get [Dockerfile](https://github.com/micro-ROS/micro_ros_zephyr_module/files/9315303/Dockerfile.txt)
```bash
docker build -t zephyr-build:jammy .
```
## Run
```bash
docker run -ti --rm zephyr-build:jammy
```

## Inside container

```bash
git clone -b humble https://github.com/micro-ROS/micro_ros_zephyr_module.git
west build -b disco_l475_iot1 micro_ros_zephyr_module
```

Outcome:
```text
(...)

Memory region         Used Size  Region Size  %age Used
           FLASH:      130299 B         1 MB     12.43%
            SRAM:       57536 B        96 KB     58.53%
        IDT_LIST:          0 GB         2 KB      0.00%
```

closed 2022-08-12 by pablogs9

acastro4, 2023-02-23:
Hello, @robodrome what are the next steps to flash micro-ros to the board?

Thanks

robodrome, 2023-02-23:
@acastro4 

That would be:
```bash
west flash --build-dir build/microros
```

See edit in my previous post.

robodrome, 2023-02-27:
> @robodrome, After runinng: `west flash --build-dir build/microros` Output: `FATAL ERROR: no CMake cache found (expected one at build/microros/CMakeCache.txt)`
> 
> EDIT: It works with a few changes
> 
> Install openocd: `sudo apt install openocd`
> 
> Then run: `west flash`

I now see the Dockerfile I now use is different from the one I posted. Well caught! And good to here it works now. I will post my Dockerfile [here](https://github.com/micro-ROS/micro_ros_zephyr_module/files/10837018/Dockerfile.txt).