micro-ROS/micro_ros_zephyr_module #79: A few build questions

gseqBE, 2022-09-20:
Hello,
I'm more familiar with the old style of using zephyr(micro_ros_setup). My issues was here, and have been directed [here](https://github.com/micro-ROS/micro_ros_setup/issues/579).

Questions:
1. Using the nightly workflow as a template, I create a docker image based off Ubuntu 22.04, with necessary apt installs. Once built, under a container, I have a build script(z.sh) which gets sourced(source z.sh), with necessary zephyr sdk install. Any guidance/comments on this setup?
[z.txt](https://github.com/micro-ROS/micro_ros_zephyr_module/files/9603320/z.txt)

2. Once the above script finishes, I clone micro_ros_zephyr_module under zephyrproject/zephyr, then build  using
`west build -p always -b disco_l475_iot1 micro_ros_zephyr_module`
but building with any other board fails:
`west build -p always -b nucleo_h743zi micro_ros_zephyr_module`
with alot of redifinition errors(see redefinition.log). There is another [issue](https://github.com/micro-ROS/micro_ros_zephyr_module/issues/28) detailing this, but I'm on v3.1 and sdk 0.14.2. I made sure to checkout v3.1 of zephyr. 
[redefinition.log](https://github.com/micro-ROS/micro_ros_zephyr_module/files/9603214/redefinition.log)

3.In the old method, one could change the transport. We are using udp. How does one do that here and configure microros. Is that required now?

- Hardware description: 
Nucleo H743zi
- RTOS: 
Zephyr
- Installation type: 
micro_ros_zephyr_module
- Version or commit hash: 
humble

#### Steps to reproduce the issue

#### Expected behavior

#### Actual behavior

#### Additional information

pablogs9, 2022-09-20:
1. I recommend you use the Zephyr Getting Started instead of our CI: https://docs.zephyrproject.org/3.1.0/develop/getting_started/index.html make sure that you check out version v3.1

2. If you source your Zephyr installation it is not necessary to clone this repo inside the Zephyr installation folder. Regarding the redefinitions, make sure that [the conf is correct](https://github.com/micro-ROS/micro_ros_zephyr_module/blob/humble/prj.conf) and also that you [have here the specific board configuration](https://github.com/micro-ROS/micro_ros_zephyr_module/tree/humble/boards). In any case, have you tried to build a Zephyr app **without micro-ROS**, and the same configuration and include `zephyr/posix/time.h` where seems to be the `error: conflicting types for 'APSR_Type'`?

3. You can change the transport by selecting in [menuconfig which one is needed to be built](https://github.com/micro-ROS/micro_ros_zephyr_module/blob/7e07006cad1d8b3170adef5d9a6642f5017021db/modules/libmicroros/CMakeLists.txt#L79-L87), [here you have the implementations](https://github.com/micro-ROS/micro_ros_zephyr_module/tree/humble/modules/libmicroros/microros_transports), and [here the transport configuration at app level](https://github.com/micro-ROS/micro_ros_zephyr_module/blob/7e07006cad1d8b3170adef5d9a6642f5017021db/src/main.c#L34-L41).

gseqBE, 2022-09-20:
Thanks.
Regarding 2.
I'm able to build successfully a blinky with `west build -p always -b nucleo_h743zi samples/basic/blinky` with time.h included in blinky/src/main.c.
Investigating...

pablogs9, 2022-09-20:
Ok, let us know what you find

gseqBE, 2022-09-20:
How would one add the board configuration to microros?
 zephyr/boards/arm has the nucleo_h743zi

pablogs9, 2022-09-21:
2. That's Zephyr configuration using .conf and .overlay files
4. This is a sample app, just modify [main.c](https://github.com/micro-ROS/micro_ros_zephyr_module/blob/humble/src/main.c)

gseqBE, 2022-09-22:
I managed to get a successful compile with the nucleo_h743zi. 
Just had to create the overlay file in the micro_ros_zephyr_module root.

`/ {
	aliases {
		uros-serial-port = &usart3;
	};
};`

I still have a question on 4. above. There was a way to separate the microros install from the application.

 In this case, the app code is under the micro_ros_zephyr_module. We have our entire application checked in a separate repository. Any suggestions on how to handle this situation?
I feel we will have to use micro_ros_zephyr_module as a submodule to our source.

Thank you.

closed 2022-09-22 by gseqBE