micro-ROS/micro_ros_zephyr_module #80: Has someone tried UDP on STM32?

Bamelink, 2022-09-21:
Hi just some general questions.

I am trying to use this module on a custom-made STM32 board we would use in Formula Student in our Driverless racing part.
I have tried several things to get it to compile with UDP as a transport layer, but when not using native posix, I get a lot of errors.
Many things are then defined twice or more if I use the socket.h with the ETH_STM32_HAL. I have attached an example of this below.

If I comment out the unistd.h, I get a step further but then it happens with other includes like the poll.h or netdb.h

I am a bit new to Zephyr and especially to ethernet. Maybe someone has an example prj.conf for example for the olimex e407? 
My other option would be to use the microros setup with the olimex e407 board and change the device tree before building, so not a high priority, but an example would fit very nicely in this repo I think.

Thanks in advance for any suggestions :)

In file included from /home/jan/micro_ros_zephyr_module/modules/libmicroros/microros_transports/udp/microros_transports.h:21,
                 from /home/jan/micro_ros_zephyr_module/src/main.c:15:
/home/jan/zephyrproject/zephyr/include/zephyr/net/socket.h:698:19: error: static declaration of 'close' follows non-static declaration
  698 | static inline int close(int sock)
      |                   ^~~~~
In file included from /home/jan/zephyr-sdk-0.14.2/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/unistd.h:4,
                 from /home/jan/micro_ros_zephyr_module/modules/libmicroros/microros_transports/udp/microros_transports.h:18,
                 from /home/jan/micro_ros_zephyr_module/src/main.c:15:
/home/jan/zephyr-sdk-0.14.2/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/unistd.h:30:9: note: previous declaration of 'close' was here
   30 | int     close (int __fildes);
      |         ^~~~~
In file included from /home/jan/micro_ros_zephyr_module/modules/libmicroros/microros_transports/udp/microros_transports.h:22,
                 from /home/jan/micro_ros_zephyr_module/src/main.c:15:
/home/jan/zephyrproject/zephyr/include/zephyr/posix/poll.h:23:19: error: redefinition of 'poll'
   23 | static inline int poll(struct pollfd *fds, int nfds, int timeout)
      |                   ^~~~
In file included from /home/jan/micro_ros_zephyr_module/modules/libmicroros/microros_transports/udp/microros_transports.h:21,
                 from /home/jan/micro_ros_zephyr_module/src/main.c:15:
/home/jan/zephyrproject/zephyr/include/zephyr/net/socket.h:775:19: note: previous definition of 'poll' was here
  775 | static inline int poll(struct zsock_pollfd *fds, int nfds, int timeout)
      |                   ^~~~

pablogs9, 2022-09-21:
Can you share the transports that you are using and some reference project for replicating this issue?

Bamelink, 2022-09-21:
I'm sorry, it was actually my bad.
I changed my prj.conf and now I can compile and link successfully.
You can find everything here: https://github.com/Bamelink/Zephyr_STM32F407
It still causes some linker errors with sys_rand with my custom board, but compiling for the olimex_stm32_e407 works

I haven't tested it on my board yet. Sorry for the unnecessary issue

pablogs9, 2022-09-21:
Ok @Bamelink, I'm closing since solved. 

We are interested in the project you are developing within Formula Student. Would you mind writing me an email with some details about the status of the project, or maybe we can have a quick meeting in order to check if we can support you somehow: pablogarrido@eprosima.com

closed 2022-09-21 by pablogs9