micro-ROS/micro_ros_zephyr_module #98: Build fails when adding CONFIG_CPLUSPLUS to prj.conf

gseqBE, 2022-10-12:
Hello,
Our code base is cpp based and works well with the non-module style of microros on zephyr.
However, with this repo(module based), I get build errors when the following two flags are added to prj.conf:
If the below two flags are removed, the default project compiles ok.
```
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
```

- Hardware description: 
nucleo_h743zi
- RTOS: 
- Installation type: 
- Version or commit hash: 
Humble

#### Steps to reproduce the issue

#### Expected behavior
Build succeeds.

#### Actual behavior
Build failure
```
/ws/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: ../modules/libmicroros/libmicroros.a(libtest_msgs__rosidl_generator_c-arrays__functions.c.obj): in function `test_msgs__srv__Arrays_Request__init':
/ws/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/build/test_msgs/rosidl_generator_c/test_msgs/srv/detail/arrays__functions.c:52: undefined reference to `test_msgs__msg__BasicTypes__init
```

[build_log.txt](https://github.com/micro-ROS/micro_ros_zephyr_module/files/9766167/build_log.txt)

#### Additional information

gseqBE, 2022-10-17:
@Acuadros95 is this something that could be fixed. Thanks!

Acuadros95, 2022-10-20:
Hi @gseqBE,

For some reason, the generated code for the `test_msgs` package does not include the `extern "C"` guards to fix name mangling on c++. 

We will take a closer look at this issue, but as those messages are used only for testing purposes, we can just leave them out of our build process for now. Please try to build again with this PR branch: https://github.com/micro-ROS/micro_ros_zephyr_module/pull/99

gseqBE, 2022-10-20:
Thanks @Acuadros95 . I still seem to get the build errors in fix/cpp_build. Were you able to try with 
```
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
```

in prj.conf?

Acuadros95, 2022-10-21:
> Were you able to try with
Yes. Which errors?

gseqBE, 2022-10-21:
Adding the build log. All I did was add the two cpp config lines in prf.conf.
[build.log](https://github.com/micro-ROS/micro_ros_zephyr_module/files/9843058/build.log)

Acuadros95, 2022-10-24:
Well, `test_msgs` is still being build on your log. Can you try on a clean environment with the `fix/cpp_build` branch?

gseqBE, 2022-10-25:
@Acuadros95  I'm  away on a work trip, please give me till this Friday to test. Thank you.

gseqBE, 2022-10-25:
@Acuadros95 
Confirming that humble branch works for me, with the two zephyr CPP flags.
Note that my board is a h743zi: `west build  -b nucleo_h743zi .`

For the default `west build -b disco_l475_iot1 .` i get an SRAM error with the flags, but someone with those flags will be using a higher SRAM chip anyway.
`ws/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `noinit' will not fit in region `SRAM'`

Thanks for the help. Issue can close.

closed 2022-10-26 by Acuadros95