micro-ROS/micro_ros_zephyr_module #103: POSIX timers for zephyr fix in rcutils is not being reflected in the humble release

uLipe, 2022-10-26:
## Issue template

- RTOS: Zephyr
- Installation type: module
- Version or commit hash: humble
- 
#### Steps to reproduce the issue
After cloning a fresh copy of micro ROS module for zephyr I was not able to build because problems with posix time implementation for Zephyr. It seems the current humble release is not reflecting the fix that was submitted to the rcutils: https://github.com/ros2/rcutils/pull/390

#### Expected behavior
It shoud build the binary to be flashed to the board using west flash

#### Actual behavior
It fails during the build, pointing error in the posix/time.h path for zephyr.

uLipe, 2022-10-26:
@pablogs9 @Acuadros95 could you please clarify?

Looking at the humble branch on the rcutils, it seems the additions for handling Zephyr posix time are not there:

https://github.com/ros2/rcutils/blob/6724d8a0f842e13cac07f2e90a06f79ae6fcfc48/src/time_unix.c#L38

uLipe, 2022-10-26:
@microbuilder, you might be interested on this.

pablogs9, 2022-10-27:
@uLipe https://github.com/ros2/rcutils/pull/390 targets `rolling` branch has not been backported to `humble`, in any case I backported manually to the micro-ROS branch.

Here you can see the `origin/ros2` branch and the micro-ROS branch above:

[image]

For me is easier to manually backport to micro-ROS but ideally, changes shall be backported in ROS 2 package. Could you take a look at the git tree and tell me which PR should I backport?

uLipe, 2022-10-27:
@pablogs9 thanks for the clarification, to get the microROS working fine across the Zephyr versions we discussed previously the PR I flagged above is required, would you be able to backport it to humble?

pablogs9, 2022-10-27:
Yes but please, let me know which commit do you need to backport to humble branch

uLipe, 2022-10-28:
@pablogs9, sure the missing commit I see is:

4767778eba04f11eaa4a924648ff4bca5433396c

pablogs9, 2022-10-28:
[image]

Let me know if this works for you

uLipe, 2022-10-28:
@pablogs9 , this works fine to me.

Thank you for your support.

pablogs9, 2022-10-28:
Cool. Closing, reopen if not solved.

closed 2022-10-28 by pablogs9