micro-ROS/micro_ros_zephyr_module #105: rcutils/src/time_unix.c fails on foxy

Bamelink, 2022-11-11:
## Time_unix.c fails

- Hardware: STM32F4_Discovery
- RTOS: Zephyr
- Installation type:  module
- Version: foxy

#### Steps to reproduce the issue
I have taken the folder "modules/libmicroros" from the foxy branch and added them to my application. When building, the following output happens:
```c
/home/jan/zephyrproject/autonomous-system-controller/Software/application/modules/libmicroros/micro_ros_src/src/rcutils/src/time_unix.c:33:10: fatal error: posix/time.h: No such file or directory
   33 | #include <posix/time.h>  //  Points to Zephyr toolchain posix time implementation
      |          ^~~~~~~~~~~~~~
```
There the include path for Zephyr is wrong (at least for my tested versions 3.1.0 and 3.2.0). Is has to be
```c
#include <zephyr/posix/time.h>
```
Changing this makes it compile and link without any error.

pablogs9, 2022-11-14:
COuld you test again from a clean environment? Ensure that rcutils package is downloaded again.

I have just backported https://github.com/ros2/rcutils/pull/390 to `foxy`.

Bamelink, 2022-11-14:
It seems to only be applied to the "rolling" branch right?

If I rerun it, it clones branch foxy and there these changes do not apply and so it still gives me the same error. Porting these changes manually I can confirm it is working

pablogs9, 2022-11-14:
Be careful because we at micro-ROS use a fork: https://github.com/micro-ROS/rcutils/tree/foxy

pablogs9, 2022-11-14:
Check the last commit in foxy branch. It is a cherry-pick of the commit that fixes your issue.

Bamelink, 2022-11-14:
Then I guess I am missing something.
But for me the foxy branch has the #if defined(__ZEPHYR__), but is missing the #if ZEPHYR_VERSION_CODE >= ZEPHYR_VERSION(3, 1, 0).
I tried both branches, humble and foxy, in a clean environment and foxy fails with the posix/time.h error while humble builds just fine

pablogs9, 2022-11-14:
... Sorry ... I did the backport locally and I did not push the changes...

Here you have: https://github.com/micro-ROS/rcutils/blob/foxy/src/time_unix.c

Bamelink, 2022-11-14:
Can confirm, it is working now 👍🏼

closed 2022-11-14 by Bamelink