micro-ROS/micro_ros_zephyr_module #106: std_srvs location

gseqBE, 2022-11-30:
We are moving towards zephyr microros module from the  non-module style.

There are a couple of services we are using, specifically trigger style service.
There used to be a std_srvs/srv under /mcu_ws/install/include/std_srvs/ in the non-module style and using it as below seemed to work.

[image]

I'm unable to find the std services in the module solution. Has it moved?
This is what my modules/libmicroros/include folder looks like:
[image]

Thank you!

pablogs9, 2022-12-01:
Here you have the whole list of packages that micro-ROS uses for its building, including msg and srv packages: https://github.com/micro-ROS/micro_ros_zephyr_module/blob/humble/modules/libmicroros/libmicroros.mk#L75-L94

If your required package is not available, please add it manually.

Also, it would be nice to have a way to include "user packages" as in [other modules](https://github.com/micro-ROS/micro_ros_espidf_component/tree/humble/extra_packages).

gseqBE, 2022-12-02:
If I comment out https://github.com/micro-ROS/micro_ros_zephyr_module/blob/humble/modules/libmicroros/libmicroros.mk#L98

or remove the COLCON_IGNORE from: src/common_interfaces/std_srvs folder, so it is not under separate package, but under common interfaces. Is it possible to allow it by default?

pablogs9, 2022-12-05:
Please open a PR with your suggestions.

closed 2022-12-05 by pablogs9