micro-ROS/micro_ros_zephyr_module #118: Cannot create more than 2 services

rcolatobe, 2023-01-12:
## Issue template

- Hardware description: STM32_h743vit MCU
- RTOS: Zephyr 3.1
- Installation type: MicroROS zephyr module
- Version or commit hash: Humble

#### Steps to reproduce the issue
[image]
Using the above configuration, I am trying to create and use 4 services that use trigger std messages.
All 4 services have been added to the executor, but for the last 2 services rclc_service_init_best_effort returns RCL_RET_ERROR. 

#### Expected behavior
Using the CLI on a computer connected to the ROS network
ros2 service call [tab autocomplete] displays 4 services.
Also service call CLI commands do not get a response.

#### Actual behavior
Using the CLI on a computer connected to the ROS network
ros2 service call [tab autocomplete] displays 2 services.
Also service call CLI commands do get a response.

#### Additional information

pablogs9, 2023-01-13:
Can you paste here the output of the micro-ROS Agent with flag `-v6` please?

pablogs9, 2023-01-13:
An idea:
- You are setting the number of servers here: https://github.com/micro-ROS/micro_ros_zephyr_module/blob/9395fc504f1c94db2a5fc2b24a448749862322ce/modules/libmicroros/Kconfig#L84
- This lines set the configured_colcon.meta during library build time here: https://github.com/micro-ROS/micro_ros_zephyr_module/blob/9395fc504f1c94db2a5fc2b24a448749862322ce/modules/libmicroros/libmicroros.mk#L39
- Finally, it is used to build the library.

I'm not sure what happens when you build the library, then modify the param, and the rebuild the app. So, have you tried to clean the library and configure your number of servers? Maybe starting with a clean micro-ROS environment with no generated library?

rcolatobe, 2023-01-13:
Deleting the zephyr module and pulling it again to refresh did fix the issue.

Is there a way to clean build the micro_ros_zephyr_module so that I don't delete the entire folder?

pablogs9, 2023-01-16:
No, I guess that there is not a cleaner way. If you find some way, please consider opening a PR.

I'm closing as resolved.

closed 2023-01-16 by pablogs9