micro-ROS/micro_ros_zephyr_module #119: How does someone make use of this in a zephyr project?

chmorgan, 2023-04-24:
How should someone add this module to their existing zephyr project?

pablogs9, 2023-04-24:
This is a sample application, we will modify it to be a pure module. But by now, you can replicate this structure in you zephyr project.

gseqBE, 2023-07-28:
@chmorgan 
I had asked this question [some time ago](https://github.com/micro-ROS/micro_ros_zephyr_module/issues/81)

1. Treated this repo as a submodule
2. We placed it in a folder called libraries, where all our external submodules are.
3. Add to Cmakelists.txt
`list(APPEND ZEPHYR_EXTRA_MODULES ${CMAKE_CURRENT_SOURCE_DIR}/<library path from above>/micro_ros_zephyr_module/modules/libmicroros/)`
5. We also had the below in the project .conf
`# Microros
CONFIG_MICROROS=y
CONFIG_APP_LINK_WITH_MICROROS=y`
7. Then: `west build -t menuconfig, enable libmicroros under modules and make config changes.`

CRichterLeipzig, 2025-05-28:
If i do so, i become:

```
`*  Executing task: west build -t menuconfig /home/ros2/zephyr/workspace/blinky_pwm --build-dir /home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0  

-- west build: running target menuconfig
[0/1] cd /home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/c...cripts/kconfig/menuconfig.py /home/ros2/zephyr/workspace/external/zephyr/Kconfig
Loaded configuration '/home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/zephyr/.config'
Configuration saved to '/home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/zephyr/.config'
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task: west build /home/ros2/zephyr/workspace/blinky_pwm --build-dir /home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0  

[0/1] Re-running CMake...
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/ros2/zephyr/workspace/blinky_pwm
-- CMake version: 3.28.3
-- Cache files will be written to: /home/ros2/.cache/zephyr
-- Zephyr version: 4.1.0 (/home/ros2/zephyr/workspace/external/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: lpcxpresso55s69, qualifiers: lpc55s69/cpu0
-- Found host-tools: zephyr 0.17.0 (/home/ros2/.zephyr_ide/toolchains/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/home/ros2/.zephyr_ide/toolchains/zephyr-sdk-0.17.0)
-- Found BOARD.dts: /home/ros2/zephyr/workspace/external/zephyr/boards/nxp/lpcxpresso55s69/lpcxpresso55s69_lpc55s69_cpu0.dts
-- Generated zephyr.dts: /home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/zephyr/zephyr.dts
-- Generated pickled edt: /home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/zephyr/edt.pickle
-- Generated devicetree_generated.h: /home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/zephyr/dts.cmake
Parsing /home/ros2/zephyr/workspace/external/zephyr/Kconfig
Loaded configuration '/home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/zephyr/.config'
No change to configuration in '/home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/zephyr/.config'
Kconfig header saved to '/home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/zephyr/include/generated/zephyr/autoconf.h'
Load components for LPC55S69_cm33_core0:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_lpc_gpio component is included.
driver_pint component is included.
driver_inputmux component is included.
driver_inputmux_connections component is included.
driver_sctimer component is included.
driver_flexcomm component is included.
driver_flexcomm_usart component is included.
driver_lpc_iocon component is included.
-- Using ccache: /usr/bin/ccache
-- Configuring done (2.5s)
-- Generating done (0.0s)
-- Build files have been written to: /home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0
[143/156] Performing build step for 'libmicroros_project'
FAILED: libmicroros-prefix/src/libmicroros_project-stamp/libmicroros_project-build /home/ros2/zephyr/workspace/modules/micro_ros_zephyr_module/modules/libmicroros/libmicroros.a /home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/libmicroros-prefix/src/libmicroros_project-stamp/libmicroros_project-build 
cd /home/ros2/zephyr/workspace/modules/micro_ros_zephyr_module/modules/libmicroros && make -f libmicroros.mk "X_CFLAGS=-I/home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/zephyr/include/generated/zephyr -I/home/ros2/zephyr/workspace/external/zephyr/include -I/home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0/zephyr/include/generated -I/home/ros2/zephyr/workspace/external/zephyr/soc/nxp/lpc -I/home/ros2/zephyr/workspace/external/zephyr/lib/libc/common/include -I/home/ros2/zephyr/workspace/external/zephyr/soc/nxp/lpc/lpc55xxx/. -I/home/ros2/zephyr/workspace/external/modules/hal/cmsis/CMSIS/Core/Include -I/home/ros2/zephyr/workspace/external/zephyr/modules/cmsis/. -I/home/ros2/zephyr/workspace/external/modules/hal/nxp/mcux/mcux-sdk/devices/LPC55S69 -I/home/ros2/zephyr/workspace/external/modules/hal/nxp/mcux/mcux-sdk/devices/LPC55S69/drivers -I/home/ros2/zephyr/workspace/external/modules/hal/nxp/mcux/mcux-sdk/drivers/common -I/home/ros2/zephyr/workspace/external/modules/hal/nxp/mcux/mcux-sdk/drivers/lpc_gpio -I/home/ros2/zephyr/workspace/external/modules/hal/nxp/mcux/mcux-sdk/drivers/pint -I/home/ros2/zephyr/workspace/external/modules/hal/nxp/mcux/mcux-sdk/drivers/inputmux -I/home/ros2/zephyr/workspace/external/modules/hal/nxp/mcux/mcux-sdk/drivers/sctimer -I/home/ros2/zephyr/workspace/external/modules/hal/nxp/mcux/mcux-sdk/drivers/flexcomm -I/home/ros2/zephyr/workspace/external/modules/hal/nxp/mcux/mcux-sdk/drivers/flexcomm/usart -I/home/ros2/zephyr/workspace/external/modules/hal/nxp/mcux/mcux-sdk/drivers/lpc_iocon -I/home/ros2/zephyr/workspace/external/zephyr/modules/hal_nxp/. -DKERNEL -D__ZEPHYR__=1 -D__LINUX_ERRNO_EXTENSIONS__ -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__PROGRAM_START -DCPU_LPC55S69JBD100_cm33_core0 -DNDEBUG -f [... cut ...]
make: *** [libmicroros.mk:108: /home/ros2/zephyr/workspace/modules/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install] Fehler 2
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0

 *  The terminal process "/usr/bin/bash '-c', 'west build /home/ros2/zephyr/workspace/blinky_pwm --build-dir /home/ros2/zephyr/workspace/blinky_pwm/build/lpcxpresso55s69/lpc55s69/cpu0 '" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. `
```