micro-ROS/micro_ros_zephyr_module #131: Support for chip from espressif family with Zephyr

OgnjenX, 2023-11-27:
Is it possible to use this module on an ESP chip (I'm trying to use it on ESP32)?
I'm able to build 'micro_ros_zephyr_module' using 'west build -b disco_l475_iot1' but I'm having errors caused by 'app.overlay' content when building with 'west build -p auto -b esp32_devkitc_wroom':

-- west build: generating a build systemwroom
Loading Zephyr default modules (Zephyr base).
-- Application: /home/ognjen/zephyrproject/applications/micRobbit-firmware/app/modules/micro_ros_zephyr_module
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/ognjen/.cache/zephyr
-- Zephyr version: 3.5.99 (/home/ognjen/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32_devkitc_wroom
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
-- Found toolchain: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
-- Found Dtc: /opt/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/ognjen/zephyrproject/zephyr/boards/xtensa/esp32_devkitc_wroom/esp32_devkitc_wroom.dts
-- Found devicetree overlay: /home/ognjen/zephyrproject/applications/micRobbit-firmware/app/modules/micro_ros_zephyr_module/app.overlay
devicetree error: /home/ognjen/zephyrproject/applications/micRobbit-firmware/app/modules/micro_ros_zephyr_module/app.overlay:2 (column 14): parse error: undefined node label 'usbotg_fs'
-- In: /home/ognjen/zephyrproject/applications/micRobbit-firmware/app/modules/micro_ros_zephyr_module/build/zephyr, command: /usr/bin/python3;/home/ognjen/zephyrproject/zephyr/scripts/dts/gen_defines.py;--dts;/home/ognjen/zephyrproject/applications/micRobbit-firmware/app/modules/micro_ros_zephyr_module/build/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/home/ognjen/zephyrproject/zephyr/dts/bindings;--header-out;/home/ognjen/zephyrproject/applications/micRobbit-firmware/app/modules/micro_ros_zephyr_module/build/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/home/ognjen/zephyrproject/applications/micRobbit-firmware/app/modules/micro_ros_zephyr_module/build/zephyr/zephyr.dts.new;--edt-pickle-out;/home/ognjen/zephyrproject/applications/micRobbit-firmware/app/modules/micro_ros_zephyr_module/build/zephyr/edt.pickle;--vendor-prefixes;/home/ognjen/zephyrproject/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at /home/ognjen/zephyrproject/zephyr/cmake/modules/dts.cmake:268 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /home/ognjen/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /home/ognjen/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/ognjen/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/ognjen/zephyrproject/applications/micRobbit-firmware/app/modules/micro_ros_zephyr_module/build -GNinja -DBOARD=esp32_devkitc_wroom -S/home/ognjen/zephyrproject/applications/micRobbit-firmware/app/modules/micro_ros_zephyr_module

I found in this issue statement that this module should be usable on any board supported by Zephyr.

https://github.com/micro-ROS/micro_ros_zephyr_module/issues/60#issuecomment-1141161320

Thanks in advance.

pablogs9, 2023-11-28:
Does it works if you revert this changes: https://github.com/micro-ROS/micro_ros_zephyr_module/pull/125 ?

OgnjenX, 2023-11-28:
Thanks for the fast response @pablogs9.
If I revert the whole commit, I have problems with undefined references to ring buffer related functions... But if I just remove the 'app.overlay' build works.
I managed to build my project with the microROS library by just linking 'libmicroros' (https://github.com/micro-ROS/micro_ros_zephyr_module/tree/iron/modules/libmicroros), so I'm not sure if that is the way to go? (and if yes, then building 'micro_ros_zephyr_module' as a whole is probably less relevant?)

pablogs9, 2023-11-28:
This repo is an example of a micro-ROS application for `disco_l475_iot1`, in that sense, you will need to tune the main folder to fit your platform and "reuse" the module (https://github.com/micro-ROS/micro_ros_zephyr_module/tree/iron/modules/libmicroros).

Let me know if you have any questions.

pablogs9, 2023-12-04:
I do not know how big is your flash. CPP std lib tends to be big and micro-ROS is also a big library (depending on the types and features you use). Normally is a good idea to build with -Os and with -ffuntions -fsections to allow the linker to optimize out not used code sections.

OgnjenX, 2023-12-07:
@towen I believe we are experiencing the same issue... For some reason it breaks when `rcl_support_init` is called (concretely it breaks on line `srand(uxr_nanos());` - I wondered if newlibc in espressif toolchain is missing).
Any help would be appreciated.
Thanks.

OgnjenX, 2023-12-09:
I tried replacing srand and rand, but there is an issue with all libc functions. I'm just wondering if that is happening on other targets or if that is a problem specific to espressif's toolchain in Zephyr...

OgnjenX, 2023-12-09:
Replacing POSIX functions with Zephyr specifics for random number generation would require modifying 'rmw-microxrcedds' source code (which, I think, is not an ideal option - I think it is using POSIX APIs for easier portability on different platforms).
Assuming that the project example contained in this repo works on the 'disco_l475_iot' board (which I cannot check because I do not have one), that would mean that there is a problem with espressif toolchain in Zephyr (it could probably mean something else, but I do not have better explanation currently in mind).

sylvioalves, 2023-12-11:
@OgnjenX, while I don't have a proper fix for this issue yet, please, would you try commenting this line?  
https://github.com/zephyrproject-rtos/hal_espressif/blob/zephyr/zephyr/esp32/CMakeLists.txt#L77  

Let me know if it solves it for now.

OgnjenX, 2023-12-11:
@sylvioalves That solves the issue with newlib functions used in the app, so I guess it is a reasonable workaround until the issue is fixed.
Thank you very much!

OgnjenX, 2023-12-11:
There is still the issue with `rclc_support_init`, it always returns `RMW_RET_ERROR` (in my application, as well as in a sample app from this repo) @pablogs9 

Any help would be appreciated.

nilskiefer, 2025-07-17:
Any progress on this? I am also seeing `RMW_RET_ERROR` from `rclc_support_init`