micro-ROS/micro_ros_zephyr_module #132: Pip dependencies error while running Zephyr Emulator tutorial

Arnix39, 2023-12-13:
I am trying to run the Zephyr emulator tutorial in a VS code's dev container using a ros:humble image.
Steps to reproduce the issue

I cloned the Micro-ROS setup repo and launched the dev container which is running rosdep and building the initial workspace.
Expected behavior

Create step is effectively creating the firmware folder.
Actual behavior

The create step yields the following error (when running pip install --force-reinstall Sphinx==4.2.0 ):

    ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
    sphinx-tabs 3.2.0 requires docutils~=0.16.0, but you have docutils 0.17.1 which is incompatible.
    sphinx-rtd-theme 0.5.2 requires docutils<0.17, but you have docutils 0.17.1 which is incompatible.

Additional information

It looks like the conflicts are coming from firmware/zephyrproject/zephyr/scripts/requirements-doc.txt where specified sphinx version is 3.*

pablogs9, 2023-12-13:
Are you able to run the emulator using this Zephyr component?

Arnix39, 2023-12-13:
I managed to downgrade manually the version of docutils.
However it seems I can't start the emulator:
```
arnix@faye:/home/workspace$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for zephyr platform host
WARNING: Using a test - not safe - entropy source
*** Booting Zephyr OS build zephyr-v2.6.0  ***
Failed status on line 98: 1. Aborting.
```