micro-ROS/micro_ros_zephyr_module #133: Micro ros agent stopped at set_verbose_level

czarekk215, 2024-01-05:
## Issue template

- Hardware description: Wio LoRa E5 dev board
- Zephyr
- Version or commit hash: Humble

## Steps to reproduce the issue
I reproduce steps like on website https://micro.ros.org/docs/tutorials/core/first_application_rtos/zephyr/
I tried pin_pong app and int32_publisher but I used `serial` config
But to run it on my board I added my board name in CMakeList.txt and also changed in microro_tranpsort.h  fd:
```
    // Set the UART used for communication. e.g. {.fd = 0} leads to UART_0 being used.
    static zephyr_transport_params_t default_params = {.fd = 0};
```
to `.fd=1`.
I flash the board by st-link, but for connection I use UART-USB concerter
```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_c68cb4bd91ecec11be34ac0ead51a8b2-if00-port0
```

#### Actual behavior
```
[1612884194.563610] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1612884194.563954] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```
#### Expected bahviour
I want to have a new publisher `ping` in ros2 topics, but for now i only have 2 default. This is what I expect:
```
user@user:~$ ros2 topic echo /microROS/ping
stamp:
  sec: 20
  nanosec: 867000000
frame_id: '1344887256_1085377743'
---
stamp:
  sec: 25
  nanosec: 942000000
frame_id: '730417256_1085377743'
---
```
#### Additional information
Is there anything to add or what should I change?
I think the problem is in connection between board and my pc, I work on ubnutu 22.04.3, but I don't know why it's starts to running because if there wasn't connection it would show `waiting for connection`

czarekk215, 2024-01-08:
@pablogs9

pablogs9, 2024-01-08:
Please try to use this module instead of `micro_ros_setup`

czarekk215, 2024-01-09:
Firstly, I tried to build this module, as it is showed for disco_l475_iot, but i got this error. I have zephyr 3.1.0 and sdk 14.2
```
cezary@cezary-MacBookPro:~/zephyrproject$ west build -b disco_l475_iot1 micro_ros_zephyr_module
[1/19] Performing build step for 'libmicroros_project'
FAILED: libmicroros-prefix/src/libmicroros_project-stamp/libmicroros_project-build /home/cezary/zephyrproject/micro_ros_zephyr_module/modules/libmicroros/libmicroros.a /home/cezary/zephyrproject/build/libmicroros-prefix/src/libmicroros_project-stamp/libmicroros_project-build 
cd /home/cezary/zephyrproject/micro_ros_zephyr_module/modules/libmicroros && make -f libmicroros.mk "X_CFLAGS=-I/home/cezary/zephyrproject/zephyr/include/zephyr -I/home/cezary/zephyrproject/zephyr/include -I/home/cezary/zephyrproject/build/zephyr/include/generated -I/home/cezary/zephyrproject/zephyr/soc/arm/st_stm32/stm32l4 -I/home/cezary/zephyrproject/zephyr/lib/libc/newlib/include -I/home/cezary/zephyrproject/zephyr/drivers -I/home/cezary/zephyrproject/zephyr/soc/arm/st_stm32/common -I/home/cezary/zephyrproject/zephyr/subsys/usb/device -I/home/cezary/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/cezary/zephyrproject/modules/hal/atmel/include -I/home/cezary/zephyrproject/zephyr/modules/hal_nxp/usb/. -I/home/cezary/zephyrproject/modules/hal/stm32/stm32cube/stm32l4xx/soc -I/home/cezary/zephyrproject/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include -I/home/cezary/zephyrproject/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include/Legacy -I/home/cezary/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -DSTM32L475xx -DUSE_HAL_DRIVER -DUSE_FULL_LL_DRIVER -DCORE_CM4 -DHSE_VALUE=8000000 -Os -imacros /home/cezary/zephyrproject/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/home/cezary/zephyr-sdk-0.14.2/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/cezary/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/cezary/zephyrproject/micro_ros_zephyr_module=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/cezary/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/cezary/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections " "X_CXXFLAGS=-I/home/cezary/zephyrproject/zephyr/include/zephyr -I/home/cezary/zephyrproject/zephyr/include -I/home/cezary/zephyrproject/build/zephyr/include/generated -I/home/cezary/zephyrproject/zephyr/soc/arm/st_stm32/stm32l4 -I/home/cezary/zephyrproject/zephyr/lib/libc/newlib/include -I/home/cezary/zephyrproject/zephyr/drivers -I/home/cezary/zephyrproject/zephyr/soc/arm/st_stm32/common -I/home/cezary/zephyrproject/zephyr/subsys/usb/device -I/home/cezary/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/cezary/zephyrproject/modules/hal/atmel/include -I/home/cezary/zephyrproject/zephyr/modules/hal_nxp/usb/. -I/home/cezary/zephyrproject/modules/hal/stm32/stm32cube/stm32l4xx/soc -I/home/cezary/zephyrproject/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include -I/home/cezary/zephyrproject/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include/Legacy -I/home/cezary/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -DSTM32L475xx -DUSE_HAL_DRIVER -DUSE_FULL_LL_DRIVER -DCORE_CM4 -DHSE_VALUE=8000000 -Os -fcheck-new -fno-exceptions -fno-rtti -imacros /home/cezary/zephyrproject/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/home/cezary/zephyr-sdk-0.14.2/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/cezary/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/cezary/zephyrproject/micro_ros_zephyr_module=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/cezary/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/cezary/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections " X_CC=/home/cezary/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc X_AR=/home/cezary/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-ar X_RANLIB=/home/cezary/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-ranlib X_CXX=/home/cezary/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++ COMPONENT_PATH=/home/cezary/zephyrproject/micro_ros_zephyr_module/modules/libmicroros ZEPHYR_BASE=/home/cezary/zephyrproject/zephyr PROJECT_BINARY_DIR=/home/cezary/zephyrproject/build/zephyr && /usr/bin/cmake -E touch /home/cezary/zephyrproject/build/libmicroros-prefix/src/libmicroros_project-stamp/libmicroros_project-build
. /home/cezary/zephyrproject/micro_ros_zephyr_module/modules/libmicroros/utils.sh; \
[... cut ...]
Aborted  <<< rosidl_typesupport_interface [1.19s]
Aborted  <<< rcutils [1.60s]
Aborted  <<< rosidl_cli [1.88s]

Summary: 0 packages finished [4.88s]
  1 package failed: microcdr
  3 packages aborted: rcutils rosidl_cli rosidl_typesupport_interface
  2 packages had stderr output: microcdr rcutils
  58 packages not processed
make: *** [libmicroros.mk:103: /home/cezary/zephyrproject/micro_ros_zephyr_module/modules/libmicroros/micro_ros_src/install] Error 2
ninja: build stopped: subcommand failed.
```

pablogs9, 2024-01-09:
Try to use ARM GCC instead of the Zephyr version `arm-zephyr-eabi-gcc`. 

In any case, you can see a working procedure in the CI: [7456517565](https://github.com/micro-ROS/micro_ros_zephyr_module/actions/runs/7456517565/job/20287290626)

Those are the steps: https://github.com/micro-ROS/micro_ros_zephyr_module/blob/7c5edcdf9e0a6c6e131d9e45ad357fa0757a5584/.github/workflows/nightly.yml#L43