micro-ROS/micro_ros_zephyr_module #134: Changed Zephyr code base causes compilation errors

danielallstar, 2024-01-10:
## Issue template

- Hardware description: STM32H743ZI
- RTOS: First noticed on Zephyr 3.5.99 
- Installation type: 
- Version or commit hash: iron

#### Steps to reproduce the issue
Dear micro-ROS maintainers,

Due to a recent change in Zephyr code it is no longer possible to successfully compile micro-ROS using the latest Zephyr code. 

The notified problem is:  " in function `uxr_nanos': undefined reference to `z_impl_clock_gettime' "
The Zephyr developers have removed the `z_impl_clock_gettime' function and replaced it by the clock_gettime function. As far as I can tell this seems to only be a change in name with similar functionality, but you can better check that yourself as well.

The change happened in commit 95a22b1 in the lib/posix/clock.c file.

https://github.com/zephyrproject-rtos/zephyr/commit/95a22b12174621aeba8ca3e0e61f7c66f03202bf#diff-d6e991eb8b553cb6e536ec823096cee7c24036762da9876a91884d9eefcd1160

#### Proposed solution
As far as I can see this function is only used in the micro-ROS [Micro-XRCE-DDS-Client](https://github.com/eProsima/Micro-XRCE-DDS-Client/tree/master)/[src](https://github.com/eProsima/Micro-XRCE-DDS-Client/tree/master/src)/[c](https://github.com/eProsima/Micro-XRCE-DDS-Client/tree/master/src/c)/[util](https://github.com/eProsima/Micro-XRCE-DDS-Client/tree/master/src/c/util)/time.c file (https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/master/src/c/util/time.c)
As a temporary fix I could use a wrapper in my main file for the new function name and could get my code compiling again. 

int z_impl_clock_gettime(clockid_t clock_id, struct timespec *ts){
return clock_gettime(clock_id, ts);
}

It would however be nice to have this change incorporated in the general micro-ROS code.

I hope to have provided sufficient information.

Kind regards

pablogs9, 2024-01-10:
Latest supported version is v3.1.0 as you can see in the [readme](https://github.com/micro-ROS/micro_ros_zephyr_module?tab=readme-ov-file#micro-ros-module-for-zephyr). 

Please feel free to create a PR with a proposed solution.

danielallstar, 2024-01-12:
I created a PR with a proposed solution: https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/378

pablogs9, 2024-01-19:
Closing this as https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/378 was merged.

BTW @danielallstar are you using 3.5.99 seamlessly with this repo? Can we add support to this version without any major problems? Would you mind reporting if there are any other changes required to add official support to this version?

closed 2024-01-19 by pablogs9

pablogs9, 2024-02-06:
@danielallstar could you propose a solution for that? micro-ROS transport setter offers a "user argument" so maybe we can use it to define at the app level the USART used