micro-ROS/micro_ros_zephyr_module #137: How to add extra package?

czarekk215, 2024-01-24:
## How to add extra package?

- RTOS: Zephyr 3.1
- Installation type: micro_ros_zephyr_module
- Version or commit hash: humble

#### Steps to reproduce the issue
I try to build add two int service, but i have problem with adding extra package. I don't know how to add ` example_interfaces `, in CMakeList.txt

#### Expected behavior
Compile
#### Actual behavior
Don't compile
#### Additional information
I create service client example from zephyr_apps, but using zephyr_module

pablogs9, 2024-01-24:
Add it here: https://github.com/micro-ROS/micro_ros_zephyr_module/blob/7c5edcdf9e0a6c6e131d9e45ad357fa0757a5584/modules/libmicroros/libmicroros.mk#L96

czarekk215, 2024-01-24:
@pablogs9 ok, thanks for fast replay. I tried, but now I have new question, how to build a custom msg or srv for zephry module. Is there any way?

pablogs9, 2024-01-24:
https://github.com/micro-ROS/micro_ros_zephyr_module/issues/104